ArduPilot/ardupilot #5984: Compass Varience and Inconsistent Compasses

philromford, 2017-04-03:
Yesterday for the first time I had trouble with my Pixhawk PX4 with Ublox M8N GPS/Compass. It would not intialise. I was getting the message: 'Inconsistent Compasses'. I have tried many time to re-calibrate, however, consistently getting the same message. I then read about the M8N compass direction, so I tried changes in Yaw; this resulted in 'Compass Variance' messages. Reverting back to 'None' for compass orientation now continues to give the 'Compass Variance' message. When I first set up the Hexacopter, I had an older version of Mission Planner that worked for me at that time, about 10 months ago. 

When the problem occurred yesterday I had v3.4.3 firmware installed. I have just installed v3.4.6. This however, has made no difference. I have MP version 1.3.45

Can someone please help, as I have no idea what to do now.
Thanks in advance

Copter platform

Hex

Pixhawk PX4

https://www.dropbox.com/s/ul4r3q7ytpc8zlb/206.BIN?dl=0

https://www.dropbox.com/s/2t6uh7wd1lprk18/207.BIN?dl=0

Log 206 is of the last good flight
Log 207 of of the very last flight when the hex took off then immediately flipped over onto the ground upside down.

I'm at my wits end now and desperately need help guys!

OXINARF, 2017-04-03:
This is a support question and those should be posted in the forum (http://discuss.ardupilot.org). You'll probably get an answer there much faster than here.

closed 2017-04-03 by OXINARF

philromford, 2017-04-03:
Thank you, I forgot that. However, I have now been able to fix the problem.

mboland, 2017-04-03:
Would you mind putting your post in the forum anyway with your fix.
Or at least post the fix here?
Other might be interested in the solution.