ArduPilot/ardupilot #5989: Enable Oreo LEDS, and Solo tunes (with a twist?)

proficnc, 2017-04-04:
#### Issue details
https://github.com/ArduPilot/ardupilot/blob/d170396acf309cf05bbafef503d7f252042490d4/libraries/AP_Notify/AP_Notify.h#L25

the oreoLEDS are off by default.  we need them set by maybe adding a BRD_SOLO option.

there are more things that are specifically solo related, AHRS_Orient to 12 for example... and the solo tunes... some gimbal stuff etc... that may all benefit from this parameter. 

#### Version
copter 3.5 / Master (3.6)

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
SOLO
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk 2.0, 2.1, 2.1 Solo

@Pedals2Paddles @tridge

Pedals2Paddles, 2017-04-06:
Can we also bring these commits from Hugh Eaves into master?  He made these modifications to allow custom control of the LEDs. These commits allow the cube to accept the custom commands. He had some nice Python scripts that run on the solo (or maybe APsync) that can change the LEDs to do many other cool things. This doesn't appear to be something that would adversely affect other things. It's a pretty popular mod among the solo users today.
https://github.com/hugheaves/ardupilot-solo/commits/solo-led-control

OXINARF, 2017-07-05:
@Pedals2Paddles This is done right?

Pedals2Paddles, 2017-07-05:
Yup! Closed.

closed 2017-07-05 by Pedals2Paddles