ArduPilot/ardupilot #5990: PWM output for industrial drives

proficnc, 2017-04-04:
#### Issue details
_Please describe the problem, or desired feature_
Adding the capability to drive industrial servos directly from Ardupilot.

the desire here is to have the option of a signal that has a duty cycle from 10% to 90% with no gap between.

[image]

so for a 400Hz signal, this would be 
full reverse...  400us on, 3600us off
stopped        2000us on, 2000us off
full forward   3600us on, 400us off

at 20Hz it would be
full reverse... 2ms on 18ms off
stopped...     10ms on 10ms off
Full forward.. 18ms on 2ms off

this change allows us to control all manner of off the shelf motor drive systems.

#### Version
_What version was the issue encountered with_
Master

#### Platform
[  ] All
[X] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
The Cube, or any other Pixhawk compatible system
#### Logs
_Please provide a link to any relevant logs that show the issue_

[Clearpath-Driver.pdf](https://github.com/ArduPilot/ardupilot/files/892029/Clearpath-Driver.pdf)

proficnc, 2017-04-04:
possibly related to

https://github.com/ArduPilot/ardupilot/issues/5341

NickNunno, 2017-04-04:
The graph is a bit misleading. This particular drive will read from .1 to 99%. Absolute 1% and 100% are read as a dead band but any duty cycle between is an acceptable control range.

EShamaev, 2017-04-05:
From the same PDF.
[image]

Maybe support that one initially?

NickNunno, 2017-04-05:
Frequency input mode or unipolar pwm input mode require a dedicated pin as a directional trigger. Input High/Low = Direction A/B

EShamaev, 2017-04-05:
As far as I understand in AntennaTracker there are plenty outputs that are free?

NickNunno, 2017-04-05:
@proficnc is the lead on this project. 

I've yet to test frequency input mode on the drive but if this mode, with directional pin is easier to facilitate and maintain within rover code and has a higher degree of accuracy, I don't see why that wouldn't be a better option.

khancyr, 2017-08-25:
Done ! starting from https://github.com/ArduPilot/ardupilot/commit/d525c0bffb5a0d561dacef5eec154aed9d6ada26

closed 2017-08-25 by khancyr