ArduPilot/ardupilot #6002: Gps fix type LED flash

proficnc, 2017-04-05:
#### Issue details
_Please describe the problem, or desired feature_
Currently we have a green flashing LED that indicates a 3DFix, a faster flash indicates dgps, but now we have RTK Float and RTK Fix, it would be nice to have an indication that these modes are active.

My personal preference is for the green to stay with a purple flash for 50ms  every 1Hz for RTK float, and a double flash for RTK Fix. 

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

ThorstenDP, 2017-04-05:
1+
It would be good to have a sound as well when a new GPS status is reached. First it is good to know, because the LED is sometimes hard to see, and second, it is a security issue. I once had a jump in position of about 1.5m directly after takeoff when status switched form 3 to 4.

WickedShell, 2017-04-05:
I'm fine with more indicators, I'd like to keep the rapid green as indicating SBAS and above if possible ;)