ArduPilot/ardupilot #6009: CAM Relay Duration Parameter

mapircamera, 2017-04-06:
#### Issue details
As a relay pulse (trigger) is currently setup, it seems to last for a non-standard and non-adjustable length of time. If the user is trying to trigger a device (like a camera) that requires a certain length/duration for the relay (high-low) pulse, they require a parameter to be set to define this length of time. For instance, our camera requires a 1ms (1000us) relay pulse (high-low 0-3V) to trigger, though the relay pulse that exists now lasts for only 600us (so it can't be triggered). We are not using a standard PWM pulse because of the delay for our camera to decipher the PWM pulse and then act on it would be longer than acting on a relay pulse. This is going to become more important as camera manufactures (like ourselves) start designing cameras capable of much faster capture rates and require the best precision in trigger/exposure timing to work with PPK/RTK systems.

We propose adding a new parameter, perhaps named something like CAM_RELAY_DUR, that has a unit of microsecond (us), and can range from 1us to 1000000us (1sec).

We are also sending the same 1ms 0-3V pulse out at the sensor exposure moment and would like the CAM_FEEDBACK_PIN to support such a pulse. I believe AUX4 pin on a Pixhawk can be used to allow for the trigger time to be as short as 1ms, but it may not match exactly with our camera's exposure feedback.

#### Version
Current

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All

#### Hardware type
Pixhawk 1, Pixhawk 2, Pixracer, etc

#### Logs
N/A

proficnc, 2017-04-06:
if you set the duration to 0
it still works, but the pulse is 600-1600us
verified on my scope.

WickedShell, 2017-04-06:
Yeah 0 is undefined behavior, basically it only works because you have until the 50Hz pin off task hits.

A new parameter would be required (and deprecating the old one) because of the scale change. Supporting a 20ms resolution is easy, I suspect you want better then that though, which would require a more intense rewrite.

I'm not sure what you are expecting as the behavior on CAM_FEEDBACK_PIN to be? For a 1ms pulse to be reliably detected you would have to use the high resolution Aux4 pin.

mapircamera, 2017-04-07:
Yes, we'd prefer at least 1ms resolution, prefer 1us resolution though so future cameras, if they require higher resolution would be supported.

If the AUX4 pin is already setup to receive a 1ms high-low (0-3V) pulse, then I'll need to check if it's working already.

mapircamera, 2017-04-20:
Hello @tridge, any update on changing the relay/PWM duration to smaller unit scale? As a reminder we need a reliable 1ms relay pulse to trigger our cameras consistently. Thanks

tridge, 2017-05-05:
I have implemented it here:
https://github.com/ArduPilot/ardupilot/pull/6197
I removed CAM_DURATION and added CAM_TRIGG_TIME, as a time in seconds. You can set 0.0001 to get 100 microseconds. In my tests it seems to be reasonably accurate - asking for 200us actually gets around 220us or so. I can improve the accuracy a bit more, but it would make the code quite a bit more complex.

tridge, 2017-05-05:
@mapircamera if you can let me know what board type you use, I can make you a binary with this change so you can test it

mapircamera, 2017-05-08:
Really appreciate the change tridge. We're using ArduCoptor on Pixhawk 1 right now but will be using Pixhawk 2.1 very soon. I'm going to ask EShamaev and see if he can build a version for us with this change, since he's sending us some tweaked UAVCAN supported versions right now.

EShamaev, 2017-05-09:
Yes, I will be glad to provide you with the latest master firmware for test.

mapircamera, 2017-05-16:
Verified the build supplied by EShamaev successfully allows us to send a short enough relay pulse duration to trigger our cameras. Thanks guys for adding the new change. We look forward to it showing up in master soon.

mapircamera, 2018-05-30:
Hello, it has been a year now and we still do not see the CAM_TRIGG_TIME added in the master builds. Can you let us know when it will be added? We are not able to send a trigger time smaller than 1/10s with the current master builds, which is too slow for accurate triggering of our cameras.

@EShamaev could you send a version of the current master Arducoptor and Arduplane to me?

Thanks

Evan1010, 2018-06-03:
I, for one, am always looking to get higher capture rates.
As Ardupilot does not appear to support multiple independent cameras, this seems the most appropriate way to prepare for future (faster) imaging hardware.

Looking forward to seeing this integrated in Master.

EShamaev, 2018-06-07:
https://github.com/ArduPilot/ardupilot/pull/6197