ArduPilot/ardupilot #6029: AP_BattMonitor: SMBus Ehancements

WickedShell, 2017-04-10:
#### Issue details
This is meant to be a tracking issue for planned SMBus specific features, and the required support in AP_BattMonitor. I've had the same conversation 4 times over the last week covering the same topics with different people, so I thought it would make more sense to document everything in one place. (I'm aware of #5900, but that is solo specific).

Features:
- [ ] Fetch and report temperature (Underway, see #6020)
- [ ] Fetch cell voltages and report them (Underway, see #6020)
- [ ] Fetch battery serial number and report it (probably via a parameter, that way it is easily logged/viewed on the GCS without adding overhead to every DF message/telem message)
- [ ] Fetch the manufacturer name and date of the battery (log it to DF?)
- [ ] Fetch and utilize the BATTERY_STATUS ( 0x16) fields for throwing alarms/low battery levels
- [ ] Use the CHARGE_STATUS ( 0x0d ) in the Maxell driver rather then having the autopilot integrate current consumption. (Also fetch capacity values, rather then utilizing a parameter).
- [ ] Support cell based failsafes on the autopilot (spread or low values)
- [ ] Unify redundant code (Not all of the Maxell vs Solo stuff can easily be unified, however there are a number of things that can be, which would reduce flash consumption, lower maintenance overhead and reduce the chances of erroneous behavior.)
  - [ ] get_PEC() is entirely the same, and specified by the SMBus protocol, and should be moved into the common SMBus code.
  - [ ] read_word() can be unified (just requires an API to check if the PEC is supported or required)
  - [ ] read_block() might be possible to unify, solo vs maxell treats this a bit different as far as interfacing with the battery is concerned.
  - [ ] Create a common SMBus helper that reads common SMBus fields (voltage, current, status, temperature) and use this in the backends rather then duplicating it.

Those are just the ideas off the top of my head, I welcome other comments and requests. (This is primarily intended to be a list of common SMBus features however). I'm not promising to address all of these items immediately, but this is whats on my radar, and some of them will be addressed by myself at least soon.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

catkiller007, 2020-04-11:
Hi,

I'm designing a SMBus battery monitor with high accuracy energy accumulation  measurement ~ 0.1%.

- import and export energy.
- cycle counting
- power,current,individual cell voltage,etc.

I've noticed a limitation in the SMBus protocol:
```
Range: 0 to 32,767 mA for charge or
0 to -32,768 mA for discharge
```
The issue is that 32Amps it's a big limitation and we need to add an option to select the scaling of the current like : 32,768 ---> 327A.

Regards,
Adrian

be-ska, 2025-06-12:
@bhavilrevo no update on that PR, because some tests with 3DR and Maxell batteries was required to merge it. If you build a custom firmware with that changes on top of the latest ArduPilot stable, the scaling factors under the SpecificationInfo() block should just work.