ArduPilot/ardupilot #6034: AHRS2 does not report valid lat, lon, alt values and AHRS3 is hard coded to EKF2 values

SamuelDudley, 2017-04-11:
#### Issue details
This issue has two parts:

1. The AHRS2 MAVLink message only reports zeros for lat, lon and alt, regardless of the secondary AHRS estimator (EKFx or DCM). The problem code has been identified here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L236
As `ahrs.get_secondary_position(loc)` is not called prior to the `loc` values being used in the AHRS2 message, they are always sent as zeros.

2. The next part of the issue is that the AHRS3 MAVLink message only ever reports EKF2 values (if available). I feel that this message should report the tertiary AHRS solution (if available) instead.
For example, say I was using ArduPlane with both EKF2 and EKF3 running and EKF2 was my current current primary AHRS solution. In this case I would expect the AHRS message to contain EKF2 information, while the AHRS2 message could be either populated by EKF3 or DCM (more likely) information. The remaining AHRS solution which had not yet been reported would then be reported via the AHRS3 message.

The above example is my opinion on how I expect the AHRS messages to work. I think this issue would be a good place to discuss implementation options.

#### Version
Current master

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

amilcarlucas, 2021-01-05:
I think this is fixed now. Can you re-test?

IamPete1, 2021-08-15:
Looks like this point 1 is fixed, as for point 2 we don't send the AHRS3 message in AP at all.

closed 2021-08-15 by IamPete1