ArduPilot/ardupilot #6039: AP_Compass_HMC5843

olliw42, 2017-04-12:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
all

#### Hardware type
hmc58x3 compass

#### Logs
-

In AP_Compass_HMC5843.cpp:

Bug?: Here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_HMC5843.cpp#L448 numAttempts should be changed to num_samples, IMHO. Otherwise, if the compass isn't responding >2 times at the beginning wrong values would be digested.

Suggestion: Here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_HMC5843.cpp#L335-L363. It is documented in the datasheet that the sequence of data is x, z, y axis, and not x,y,z. To me it thus looks more natural to change the val structure to x,z,y, and to not correct for that irregular sequence later in lines 349,350.

olliw42, 2017-04-22:
added suggestions: Here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_HMC5843.cpp#L448 one may want to change, for stylistic reasons:
* numAttempts to num_attempts
* int numAttempts = 0, good_count = 0;  to uint8_t num_attempts = 0, good_count = 0;

tridge, 2017-06-16:
> Bug?: Here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_HMC5843.cpp#L448 numAttempts should be changed to num_samples, IMHO. Otherwise, if the compass isn't responding >2 times at the beginning wrong values would be digested.

I agree. Would you like to do a PR?

> Suggestion: Here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_HMC5843.cpp#L335-L363. It is documented in the datasheet that the sequence of data is x, z, y [...]

yep, agreed as well

olliw42, 2017-10-11:
I can't, sorry, I'm a git illiterate

amilcarlucas, 2017-10-22:
@tridge didn't you fix this already ? Can this be closed ?

OXINARF, 2017-10-23:
No, this isn't fixed.

olliw42, 2023-02-03:
closing it since really outdated :)

closed 2023-02-03 by olliw42