ArduPilot/ardupilot #6045: Unknown repeating tone

Georacer, 2017-04-13:
#### Issue details
Overview:
An unknown tone would play continuously from the buzzer, with no apparent reason.
Details:
In the video link (below), video 3, at time 00:54 (tlog@51.11%) the aircraft is set on the runway, ready for takeoff.
Due to I2C problems, there is bad communication with the compass and most efforts to arm fail. Eventually, the aircraft gets armed and takes off (time 08:12, tlog@65.30%).
However, since time 02:00 an unknown tone is played, which I can't find in the [tones library](http://ardupilot.org/copter/docs/common-sounds-pixhawkpx4.html)
The tones wouldn't stop until a system reboot took place and then some.

What is this tone and why was it played?

.tlog logs are attached.
.BIN logs are also attached but wont' be of much help because LOG_DISARMED:=0

#### Version
ArduPlane 3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Fixed-wing airplane

#### Hardware type
Pixhawk 1

#### Logs
Video link: https://archive.org/details/unatrap_20170411
[logs.zip](https://github.com/ArduPilot/ardupilot/files/919413/logs.zip)

rmackay9, 2017-04-14:
Probably best to raise this as a support issue in the discuss forums: http://discuss.ardupilot.org/c/arduplane

Sorry, I'm not a plane expert but I also don't recognise that tone.  I wonder if maybe it's not coming from the flight controller but instead some other piece of electronics like an ESC?

tridge, 2017-04-18:
assuming I'm listening to the right part of the video, that is the "ready to arm" "bip bip bip". It is repeating as the planes sensors are right on the edge of being OK, and its changing rapidly from being ready to arm and not being ready to arm.
There is probably a sensor (GPS HDop? compass?) that just happens to be right on the edge of the arming OK threshold

Georacer, 2017-04-18:
Writing from my phone (I'm on vacations)

This was a known bad hardware setup, with 2.2m of I2C cable for the compass, also swamping down the external compass as well.
It's very likely that this caused devices to not be responsive.

OXINARF, 2017-04-18:
@Georacer This was a support question that I let slip because it was quite complete with information. We don't do support in GitHub though, any issues should be posted in the forum. Only confirmed bugs or feature requests should be opened in this issues list.

Given that this has been answered, I'm closing it. If you have further questions, please use the forum.

closed 2017-04-18 by OXINARF