ArduPilot/ardupilot #6046: QuadPlane Pitch and Altitude Sustained Oscillation in QRTL while approaching landin point

ferdicakmak, 2017-04-13:
Hello,

When flying Quadplane in AUTO mode, aircraft swithces to QRTL after flying in RTL as fixed wing, which is fine until now. 

In QRTL mode, aircraft tries to reach Home point by pitching down (negative) the aircraft for maintaining Q_WP_SPEED = 5m/s, where the wind makes the aircraft loose altitude abruptly (causing QTUN.THR to 1) and cause pitch up and gain altitude (causing QTUN.THR to 0). Thus, while approaching to HOME point in QRTL, the aircraft moves like a dolphin. This oscillation stops when HOME point is reached and aircaft lands vertically.

[image]

I had about 10 flights like this and observed same pattern in 9 of them. Same pattern occurs even when Q_RTL_ALT = ALT_HOLD_RTL. I even tried Q_VFWD_GAIN : 0,05 with FW motor active but couldnt see any improvement. 

QTUN.ThrOut, QTUN.DCRt and QTUNCrt data from logs show that THR switches between maximum and minimum while trying to reach Home point with Q_WP_SPEED. 

[image]

You can also see oscillations in pitch angle due to THR changes.

[image]

Any suggestion to improve this behaviour is welcome.
Best Wishes,
Ferit.

#### Version
ArduPlane v7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane (QuadPlane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
PixHawk

iskess, 2017-04-14:
Excellent write up and analysis. I've been puzzled by this behavior.

magicrub, 2017-04-14:
Agreed, great write up.  Do you have logs to share?

ferdicakmak, 2017-04-14:
You can see that even when RTL alt and QRTL alt is the same, i experienced same behaviour due to an altitude difference of 4m.

[image]

I believe that the problem may be the desired climb rate downside (2 m/s) may be too much to handle for quadplane while in quad mode with a level velocity demanded (5m/s). When level velocity becomes 0, then no problem with descent rate. 

In my parameters for Quadplane, Q_WP_SPEED_DN = 150cm/s, Q_WP_SPEED_UP = 200cm/s and Q_VELZ_MAX = 200cm/s. So I believe in QRTL, flight controller uses Q_VELZ_MAX while approaching to HOME point, and Q_WP_SPEE_DN for initial landing. Thus I am planning on changing Q_VELZ_MAX = 100cm/s for the next flight.

You can find log file, parameters, kmz, and gpx files below.
[19_RAPTOR_VTOL_2_DolphinMove.zip](https://github.com/ArduPilot/ardupilot/files/921701/19_RAPTOR_VTOL_2_DolphinMove.zip)

You can also see the video of the final phase of the flight below.
[19_RAPTOR_VTOL_2_DolphinMove_Video.zip](https://github.com/ArduPilot/ardupilot/files/921713/19_RAPTOR_VTOL_2_DolphinMove_Video.zip)

iskess, 2017-07-10:
This is still happening in beta 3.8r6.

magicrub, 2017-07-10:
added to next devcall topic

ferdicakmak, 2017-07-10:
I believe such behaviour should be expected, due to contradicting response of VTOL and FW control elements in the transition manuever from FW to VTOL. I was able to minimize this behaviour by decreasing vertical speed limits.

I believe that, the firmware this can be improved by gradually increasing vertical speed limits during transition.

iskess, 2017-07-10:
Sorry I must amend my comment. The title of this issues says "sustained oscillation". I'm not seeing this but maybe it because I'm using Hybrid RTL so there isn't much distance to cover to reach the descent point. 
What I'm seeing it a big pitch up zoom climb when trying to lock on to the RTL point.  I suppose I posted in the wrong place. I'll upload a log here soon in case that helps.

iskess, 2017-07-10:
This was a VTOL_LAND. Transition starts at line 295, the pitch is ok during transition. At line 315 the pitch gets up to 41 deg. This exceeds the attitude limits set.  
[image]

There are 2 similar VTOL_LAND attempts in this .bin
https://www.dropbox.com/s/0xj06lflpyv9xmy/00000039.BIN?dl=0

expntly, 2017-07-19:
I've seen this happen a few month ago with a vehicle I built, and this also happened today with another vehicle, ArduPlane 3.8.0-beta6, as I was trying QRTL.  I've also seen somebody discuss this on discuss.ardupilot.org somewhere so this is a known bug that's been around for a while.

IamPete1, 2021-08-30:
Sorry we never got to this, I'm afraid there has been so many changes to the Q logic that this report is now of limited use.

closed 2021-08-30 by IamPete1