ArduPilot/ardupilot #6048: Dataflash IO heartbeat warning is invasive

jaxxzer, 2017-04-14:
When dataflash logging fails due to a corrupt filesystem, it spams the GCS with statustext messages.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/DataFlash_File.cpp#L189

From what I understand, this won't affect flight. Could we handle this by setting a flag for the vehicle or AP_Arming library to handle? Perhaps here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_Arming.cpp#L149

peterbarker, 2017-04-15:
On Fri, 14 Apr 2017, jaxxzer wrote:

> When dataflash logging fails due to a corrupt filesystem, it spams the GCS
> with statustext messages.
> https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/Data
> Flash_File.cpp#L189
> 
> From what I understand, this won't affect flight. Could we handle this by

It won't affect flight, but for those people that are relying on dataflash 
to be working for their mission to be successful (think CAM messages), 
this information is important.  It may affect other things on the IO 
thread - but none of those are flight-critical.

> setting a flag for the vehicle or AP_Arming library to handle? Perhaps here:https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_A
> rming.cpp#L149

There is already an arming check for this.

I could reduce the output rate to once/minute.  Or just remove the message 
altogether and people can rely on the log-sensor-health stuff instead.

Thoughts?

OXINARF, 2017-04-15:
I think that reducing it to once a minute is the best option. Not having the IO thread isn't flight critical but is important for the majority of our users.

jaxxzer, 2017-04-17:
I vote to take it out completely and let the vehicle/arming library handle it.

ongspxm, 2019-01-26:
Does this issue require any further action? adding a flag that is set when dataflash logging fails perhaps?