ArduPilot/ardupilot #6054: Auto Drift mode

raptortrias, 2017-04-15:
#### Issue details
Is it possible to setup Waypoint flight in Drift mode for arducopter.
This is for a hybrid QUAD with lifting surface that extends endurance for mapping missions.

#### Version
Any

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
QUAD

#### Hardware type
Pixhawk 1/2

#### Logs
NA

rmackay9, 2017-04-17:
I guess what you're looking for is to make sure the vehicle always points forward?  The issue is that sometimes the vehicle points towards the next waypoint and ends up flying slightly sideways?

If this is the case, in the short-term you can set the WP_YAW_BEHAVE to "3" which is to point in the direction the GPS indicates we are travelling.  This is what most helicopter users use I think.

There is also an issue raised somewhere here to address the issue of the vehicle pointing to the next waypoint "too early".  There is a good reason why it does that (i.e. it's not a bug so to speak) but we can improve it.

raptortrias, 2017-04-18:
Hi Randy,

 Thanks for suggestion, the issue is not only the YAW behaviour, the issue is that I require the copter to keep flying forwards at speed. So for example in WP mission once a WP is reached copter will YAW and flying to next waypoint. What I require is that the actual flight path resembles that of fixed wing, because if we loose airspeed and stop direction abruptly for yaw the flying surface will stall abruptly.

 Thanks

rmackay9, 2017-04-18:
Ok, I see.  Have you tried Spline waypoints?

lthall, 2018-01-12:
Issue can be solved using existing functionality. Planed functionality will further improve this so I am closing it.

closed 2018-01-12 by lthall