ArduPilot/ardupilot #6058: add intel aero flight board to automatic builder

rmackay9, 2017-04-17:
I've created a new Intel Aero RTF autopilot wiki page but we need to make it as easy as possible for people to get the best ardupilot firmware for this vehicle.  At the moment it requires them to build it for themselves but we should instead add it to our main build so they can get it from there.

wiki page: http://ardupilot.org/copter/docs/common-autopilots.html

Next step is to enable the mission planner to upload the firmware similar to how it does this for Bebop2.

toashwin, 2017-04-27:
Which firmware do I need to download to Intel Aero Platfrom?
Is it from this folder? I see four versions. Are those versions? Do I need to just flash V4?

http://firmware.us.ardupilot.org/Copter/latest/PX4/

rmackay9, 2017-04-27:
I'm afraid we don't have it being automatically being built yet.  but we will!

jagrandle, 2017-04-29:
That would be great !! We look forward to it being as simple as it currently is for the range of other platforms. Thank guys.

PR #6195 (2017-05-08): Tools: add aerofc-v1 to autobuild
I guess we need to push to test?

Fix #6058
closed 2017-05-08 by rmackay9