ArduPilot/ardupilot #6067: Checking ArduPilot with PVS-Studio static analyzer

khandeliants, 2017-04-18:
Hello,
I have found some weaknesses and bugs using PVS-Studio tool. PVS-Studio is a static code analyzer for C, C++ and C#: https://www.viva64.com/en/pvs-studio/

Analyzer warnings:
* [V773](https://www.viva64.com/en/w/V773/) ([CWE-401](https://cwe.mitre.org/data/definitions/401.html)) The function was exited without releasing the 'buffers' pointer. A memory leak is possible. [libraries/AP_HAL_Linux/VideoIn.cpp 131](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/VideoIn.cpp#L131)

  _Comment_: it is necessary to free the memory by the pointer [`buffers`](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/VideoIn.cpp#L116) using the `free` function, before exiting the `VideoIn::allocate_buffers` function in the conditional block `if (ret < 0)`.

* [V593](https://www.viva64.com/en/w/V593/) ([CWE-754](https://cwe.mitre.org/data/definitions/754.html)) Consider reviewing the expression of the 'A = B != C' kind. The expression is calculated as following: 'A = (B != C)'. [libraries/AP_HAL_Linux/Thread.cpp 175](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/Thread.cpp#L175)
  
  _Comment_: The closing bracket is placed in the wrong place. Therefore, the result of the function will be compared with 0, and either 0 or 1 will be written in `r`.

* [V762](https://www.viva64.com/en/w/V762/) It is possible a virtual function was overridden incorrectly. See qualifiers of function 'get_custom_terrain_directory' in derived class 'Util' and base class 'AP_HAL::Util'. [libraries/AP_HAL_Linux/Util.h 51](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/Util.h#L51)

  _Comment_: `get_custom_terrain_directory` function in the base class [`AP_HAL::Util`](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/Util.h#L22) has `const` qualifier, but `get_custom_terrain_directory` function in the derived class [`Util`](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/Util.h#L51) doesn't have a `const` qualifier, so the virtual override will not occur. The signatures of the functions must completely coincide.

In addition, I suggest having a look at the emails, sent from @pvs-studio.com.

Best regards,
Phillip Khandeliants

EShamaev, 2017-04-18:
First one after failure we have AP_HAL::panic so memory leak is least of evil.
Second one A = (B != C) really seems like a bug.
Third one also seems like a real bug.

OXINARF, 2017-04-18:
@khandeliants 

Hi Phillip,

Thank you for the reports. Since you have found all this, do you want to do a pull request with fixes? It would be appreciated.

> In addition, I suggest having a look at the emails, sent from @pvs-studio.com.

What does this mean? Were e-mails sent to someone?

staroselskii, 2017-04-19:
> What does this mean? Were e-mails sent to someone?

I guess I'm the one to blame. I got contacted by their marketing a couple of days ago to discuss using the product in our build process. 

I didn't really anticipate that they would be so kind to put efforts right away. 

I assume the discussion about incorporation the analyzer in the build is better to be conducted over a dev-call, though. I think this is something our project could gain from. This is an [article](https://www.viva64.com/en/b/0457/) I read about the topic.

EShamaev, 2017-04-19:
I also got contacted by them today for the same reason. Here are the contacts of a person:

Ekaterina Milovidova
www.viva64.com
em@pvs-studio.com
“PVS-Studio team”

khancyr, 2017-04-19:
I also get a mail... I already open an issue about PVS-studio some time ago but it requires some modification to be use freely.

OXINARF, 2017-04-19:
Thanks @staroselskii, @EShamaev and @khancyr. So this is a marketing stunt. They could, at least, do it honestly and refer that in the issue. According to their documentation:

> You can use PVS-Studio code analyzer for free, if you add special comments to your source code.

Yeah, no. We are not adding comments in every file in our source code to use a tool for free. They just want us to either market them with the comments or buy their product.
There are plenty of static analysers and many are free. We run one every month, it's called Coverity Scan.

Anyway, the issues reported here have now been fixed.

closed 2017-04-19 by OXINARF

staroselskii, 2017-04-19:
> So this is a marketing stunt.

I read their blog posts about how they came up with this and I personally find their reasoning to be quite sound.

 > They just want us to either market them with the comments or buy their product.

I totally get the frustration but it seems like ArduPilot as a project might gain from it. I suppose Coverity Scan didn't find these bugs or otherwise they wouldn't slip in the codebase, would they? Nobody stops us from purging the comments later on in case it doesn't pay off. I think there's something to discuss a little bit more. Even though I think ArduPilot codebase is now in a pretty good shape, there's always room to do more.

I also reckon we don't use a static analyzer on Mission Planner which has some typos and copy-paste errors and I guess PVS-Studio for C# might be worth a try.

I'm not affiliated with the PVS-Studio in any matter whatsoever. This is just my 2 cents.

OXINARF, 2017-04-19:
@staroselskii 

> I read their blog posts about how they came up with this and I personally find their reasoning to be quite sound.

My issue isn't with them doing marketing, it's with these tactics. They opened an issue like they were a regular user that happened to run this product and found these issues.
Also, from a business point-of-view, I understand the need to promote their product - there are various ways to do that though, having comments in every source code file seems exaggerated to me (it would mean we would change more than 600 files!). Specially when there are a variety of free tools that don't require anything from an open-source project.

> I suppose Coverity Scan didn't find these bugs or otherwise they wouldn't slip in the codebase, would they?

I don't know. And this actually goes against using another static analyser because the issue is that only very rarely someone looks at it. So yeah, it is possible this was already identified by Coverity and no one noticed. Even if it wasn't, there are other free tools we can use without spamming our code with marketing comments.

> I think there's something to discuss a little bit more.

Feel free to bring it to the next call, we are always open for discussion :slightly_smiling_face: 

> I also reckon we don't use a static analyzer on Mission Planner which has some typos and copy-paste errors and I guess PVS-Studio for C# might be worth a try.

That is a different repository that has its own maintainer, so his opinion can be favourable. I doubt @meee1 would want to change all source code files to include two comment lines though.

khancyr, 2017-04-19:
No need to let our files tagged ... They provide a tool to auto tag files, if you look at my previous comment, in my branch I was doing it, so it can be made for a special branch like coverty, but we also get plenty of false positive.
I think it could be good to look at its ouput from time to time

OXINARF, 2017-04-19:
@khancyr What you did is against their license. They have a section just for you :wink: :

> There was a large discussion on the topic whether it is permitted to add and remove comments in the source code so that to use the analyzer for free. We consider this method to be unacceptable, as it is a way to bypass the restrictions, imposed on a free license.

The automated tool is there to help you add the comments, but not as a temporary change, you have to commit the files after it.

khancyr, 2017-04-19:
Ho ! I am not sure this was present when I use it ^^. So nowadays I don't think it is worth it to change all headers for an analyzer...

khandeliants, 2017-04-20:
I am sorry for my long silence =)

I would like to bring my two cents to the dialogue.

It's no secret that the static analyzer helps you find bugs in the source code. However, the benefit of using static analysis is in its [regular use](https://www.viva64.com/en/b/0196/).
PVS-Studio can (and should be) used individiually by every developer on the code base, at the same time the fresh code can be checked before committing it to the repository (the crucial bugs can be gradually fixed).
Having looked through the [log](https://scan.coverity.com/projects/ardupilot-ardupilot), I found out that Coverity didn’t find the errors written in the issue. I think it is quite a nice demonstration of the analyzer abilities.

@khancyr 
Speaking about false positives: I think you’ve seen in the work with Coverity that static analyzers have such a flaw. We are working really hard to decrease the number of false positives and increase the number of useful warnings.
It was very interesting to learn about PVS-Studio false positives in ArduPilot project and we will be very grateful if you write about them in details. We could fix them by the next release.

@OXINARF 
Couple more words about the free use: at the moment we cannot make a free PVS-Studio version without any comments. Our job is not only to provide the tool, but also to continue support the users. 
As we are a small company, then it’s not beneficial for us to support such a version.
That’s why adding the comments is the “thank you” for the provided license and additional advertising of our product for the other members of the community.

If there is no wish to add the comments, you can consider purchasing a commercial license.