ArduPilot/ardupilot #6078: ArduCopter-3.4.6 has a bug when you enable px4flow

HockyYoung, 2017-04-19:
ArduCopter-3.4.6
These days I tested Ardupilot two branches, the first branch is 3.5, because the version of the optical flow part of the code changes, this part of a large change, resulting in the use of optical flow, the second branch is 3.4, this part Can support PX4Flow hover, but in the open optical flow need to add Rangefinder, this part has a serious problem, if you open the light flow, it will prompt, IMU0 has stopped aiding, IMU0 is using Optiflow, IMU1 has stopped aiding, IMU1 Is using Optiflow, this situation will lead to rest when the plane's attitude is bad, but not always the case, hope Ardupilot can give me some advice.
#### Hardware type
 @Pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_

@zlite

rmackay9, 2017-04-19:
@HockyYoung, any chance you have a dataflash log of the issue?

HockyYoung, 2017-04-19:
if the quality is 255 or higher，it works well.
I found that when the flowquality is 0, it will appear.
EKF2 IMU0 has stopped aiding
EKF2 IMU0 is using optical flow
This time the attitude is bad.
Thank you for answering me, I do not know how to give you the dataflash log

OXINARF, 2017-04-19:
@HockyYoung Please read http://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html?highlight=log#downloading-logs-via-mavlink

This is a support question and those need to be posted in the forum (http://discuss.ardupilot.org). So please open a topic there with a log attached to it (if the log is too big, you'll need to put it in a cloud service, like Dropbox or Google Drive, and share a link).

closed 2017-04-19 by OXINARF