ArduPilot/ardupilot #6088: Remove internal / external bussing on Toshiba led driver

proficnc, 2017-04-19:
#### Issue details
The LED driver is forcing a board specific behaviour on all boards
This looks for an led on the "external bus" and only if it doesn't find one will it try the internal bus.

We would like to ensure that if the LED is found on both busses, that it functions.

While discussing this issue, it was noted that some users prefer to have the internal LED turned off on the Pixhawk 1 due to a heating issue that is said to affect temperature stability of the IMU's 

So it has been suggested, that adding a parameter to selectively enable each or all busses would be good. The drivers can still be running at this stage. 

#### Version
_What version was the issue encountered with_
ALL

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
All
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk ALL especially 2

dinardip, 2017-04-19:
+1  being able to select status lights, should be an option.

IanP13, 2017-04-19:
I agree that there should be a parameter to selectively enable each or all busses rather than a default behaviour that is hard coded.

tridge, 2017-04-19:
yep, it should be on both buses, if found

lakshaydang, 2017-04-21:
when this would be available for use in the firmware ?

NickNunno, 2017-04-22:
I'd agree that this seems like a very logical implementation.

yak-54, 2017-04-30:
+1 agree 
i have this issue as i have GPS puck fitted inside a fuselage of a plane and an i have one external on the Ic2 bus and the one on the here GPS take's Priority

hdtechk, 2017-06-28:
I would really like to see  a fix for this hope someone can find way to do PR

proficnc, 2017-07-29:
@tridge Any movement on this one? 
Got quite a few people still asking for this as they are running the dual GPS.

robustini, 2017-07-29:
I'm one of this few people...

[image]

tridge, 2017-08-02:
i can fix it, yes, but not for the plane 3.8 release, as I'm finalising that already

OXINARF, 2017-09-11:
This is now supported in master (via #6910).

closed 2017-09-11 by OXINARF