ArduPilot/ardupilot #6095: mission item feedback when reach in guided mode

adrianhsm, 2017-04-20:
#### Issue details
in auto mode, when a way point reaches, we could reach an indication message, but if in guided mode, no message when a mission item is reached

#### Version
all

#### Platform
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad

#### Hardware type
pixhawk

#### Logs
no

#### ps
I have written the codes to implement this, if you think this feature should be there, I will create a pull request for it, if not, please tell me your concern as we may sometimes want to have this feature on.

OXINARF, 2017-04-20:
Related to #3416 and #6094

amilcarlucas, 2021-03-09:
Implemented in master as of today

closed 2021-03-09 by amilcarlucas