ArduPilot/ardupilot #6096: Crazy gimbal movement on turned off remote

Phil1988, 2017-04-21:
Issue details

I have a Strom32 based 3 axis gimbal connected via Mavlink to my Pixhawk.
Channels 7&8 are used to move the gimbal on pan and tilt axis.
When I turn/power on the Copter, and the Radio Transmitter is OFF, then the gimbal is moving crazy as long as the RT is turned off.
The moving is stopped as soon as I turn on the transmitter.
If I then switch it back off again then there is NO wrong behavior of the gimbal.

The Problem is, that the gimbal is hitting the camera against the frames and also turnes the cables on moving in pan axis a couple of times...

Could someone please have a look at it and solve this long time known bug?
Thanks in advance!

I am using Mission Planner 1.3.46 build 1.1.6309.37214 APM: Copter V3.4.6
Platform

[ ] All
[ ] AntennaTracker
[ x ] Copter
[ ] Plane
[ ] Rover
[ ] Sub
Airframe type

Hex
Hardware type

Pixhawk

lvale, 2017-04-21:
A few things missing
STorM32 version
Value of
http://ardupilot.org/copter/docs/parameters.html#mnt-type-mount-type
http://ardupilot.org/copter/docs/parameters.html#mnt-deflt-mode-mount-default-operating-mode
http://ardupilot.org/copter/docs/parameters.html#mnt-rc-in-roll-roll-rc-input-channel
http://ardupilot.org/copter/docs/parameters.html#mnt-rc-in-tilt-tilt-pitch-rc-input-channel
http://ardupilot.org/copter/docs/parameters.html#mnt-rc-in-pan-pan-yaw-rc-input-channel

Just a small operational note: It is considered dangerous to power on the vehicle with the radio off. Radio should be the first to be powered and the last to power off

Phil1988, 2017-04-21:
Sorry, you are right:

STorM32 v1.32 [http://www.olliw.eu/storm32bgc-wiki/STorM32_Boards](url)

MNT_TYPE: 4 (SToRM32 MAVLink)
MNT_DEFLT_MODE: 3 (RC Targeting)
MNT_RC_IN_ROLL: 0 (disabled)
MNT_RC_IN_TILT: 7 (RC7)
MNT_RC_IN_PAN: 8 (RC8)

And yes: You are right that its better to turn on the radio first and then power on, but a FC should be able to not make it "unsave" or getting this behaviour that I am faceing with the gimbal.
Sometimes people forget things ;)

If you need further infromation let me know :)

lvale, 2017-04-21:
Which firmware are you using on the STorM32?

I want to try to replicate as close as possible your issue.

Phil1988, 2017-04-21:
o323bgc-release-v096-v20160319

olliw42, 2017-04-21:
maybe I can add a comment on this: If I'm understanding it correctly, then this behavior is familiar to me. To the best of my knowledge: It is not STorM32 related (i.e. all that info on the STorM32 and parameters thing won't help much, I think it's present for all STorM32 firmware versions), it is because at startup ArduPilot under some conditions sends out faulty values. I had that resolved in betacopter code, but I haven't touched that for a year or so and honestly can't remember what the solution was, but this one could take as an indication that it might be worthwhile to look into the AP side.

lvale, 2017-04-21:
@olliw42 you might be right, that's why I'm trying to replicate the issue. :)

Phil1988, 2017-04-21:
such a VIP thread here :D 
Well I think we all know that the problem comes from the FC or better its Soft-/Firmware because the gimbal itself doesnt move if it gets no values.
If I can help you further @Ivale, let me know

lvale, 2017-04-21:
@Phil1988 I would only need a day with at least 35 hours. 24 is too short :)
I'll try to get to this later today

Phil1988, 2017-04-21:
Hahaha!
I know this feeling like from 6 days a week ;D 
I hope you are successfull in fixing this soon  - cheers! ;)

OXINARF, 2017-04-22:
@Phil1988 This is a support question and those should be posted in the forum (http://discuss.ardupilot.org), this list is for confirmed bugs and feature requests.

Since @lvale is going to test it, I'm going to leave it open for now.

lvale, 2017-04-25:
@OXINARF Issue reproduced. Going for serial debugging...

lvale, 2017-04-25:
@tridge This deserves a careful look.

I'll be getting some serial logs, and will be back.

lvale, 2017-04-25:
[Serial Saved Output2.txt](https://github.com/ArduPilot/ardupilot/files/956778/Serial.Saved.Output2.txt)
[Serial Saved Output1.txt](https://github.com/ArduPilot/ardupilot/files/956779/Serial.Saved.Output1.txt)

serial outputs without and with radio turned on

tridge, 2017-04-25:
what we need is a fix in two places. First:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Backend.cpp#L124
we need to treat input as zero if we are in RC failsafe. 
then here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Backend.cpp#L86
we need to do the same.
Unfortunately there is no hal.rcin->in_failsafe() function, which there should be. We need to add that, then use it in those two places to override the behaviour in failsafe.

Phil1988, 2017-04-26:
@OXINARF I thought its more an "issue" than a support question (cause its not me having this problem only but a bug) but either way I didnt know where else to report then here - so sorry for the mistake.

Great that @lvale and @tridge are having some effort on reproducing it and seems to find a solution :)

OXINARF, 2017-04-26:
@Phil1988 I understand your point, but we want to keep this list as clean as possible, otherwise it becomes unmanageable. This means we want issues reported here when they have been confirmed by either someone in the dev team or, at least, a few users. It has been reproduced now, so thanks for the report :slightly_smiling_face:

Phil1988, 2017-04-26:
@OXINARF : allright man ^^ I just didnt knew this.
I can understand your reasons for it. But to be honest this issue is known since a long time and reported by a lot of users and with my mistake it looks like it will be solved in the near future ;)
I will not do it again in the future also it was successfull :D

lvale, 2017-04-27:
@tridge @rmackay9 Marginally related http://discuss.ardupilot.org/t/pass-thru-default-pwm-1-9ms-when-rc-transmitter-signal-lost/16896

Phil1988, 2017-05-03:
I am trying to post a new thread in the forum, but I dont get the activation eMail.
Tried to resend the activation eMail 5 times already (started from yesterday) and nothing in my spam folder.
It would be great if the "normal way to post problems" could also work for me :D

OXINARF, 2017-05-03:
@Phil1988 what's your user?

Phil1988, 2017-05-03:
Leon333

Phil1988, 2017-05-08:
still not possible :(

OXINARF, 2017-05-09:
@Phil1988 Hi, sorry for only getting back to you now, but I was out with bad internet access. The activation e-mails aren't bouncing back so your server is getting them. I've activated your account, but please check your e-mail is correct.

Phil1988, 2017-05-11:
The emailadress was and is still correct. If you like you can send me an email manually to check that it is working (nut not the activation mail) ... 
Anyway: Thanks for activating :)

Phil1988, 2017-08-09:
This issue might be gone with the latest (3.5) firmware of arducopter.
At least I didt saw this behaviour on the last flight.

Let me check it once again tomorrow and I will report back and close this issue if its really gone.

gediminasgu, 2017-09-26:
I had the same issue and did some investigation some time ago. I think the bug is in this part of code: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Backend.cpp#L108
I think that angle min/max doesn't work. You can fix it setting proper values in your RX in FS mode. That worked for me. When your TX is not turned on, most likely your RX sends 0 on PWM channel, that's why gimbal goes crazy.
The other option is to enable joystick speed mode: http://ardupilot.org/plane/docs/parameters.html#mnt-jstick-spd-mount-joystick-speed
This option worked too even the RX is not connected, so it should work for you even with wrong FS values.

gediminasgu, 2017-09-26:
Also, it duplicates issue #2426

kantlivelong, 2018-05-24:
Same issue here

khancyr, 2019-08-02:
It should have been solve by https://github.com/ArduPilot/ardupilot/pull/9953, please reopen if not

closed 2019-08-02 by khancyr