ArduPilot/ardupilot #6098: Unable to detect board type

floaledm, 2017-04-21:
#### Issue details
In libraries/AP_BoardConfig/px4_drivers.cpp, any reason to default to an unrecognized board instead of defaulting to a Pixhawk 1? I have a Pixhawk clone (which seems to be fully functional) that keeps rebooting and sending that MAVLink message as a result...

This code change (which defaults to Pixhawk 1) allows me to use that clone: https://github.com/craftandtheory/ardupilot/commit/0ac4b3947a2a32d34a9e22318c00408dd9d2064d

If you think this is a nonissue or that you have had no one else report this, feel free to close...

#### Platform
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk 1 clone.

magicrub, 2017-04-21:
I think the main question here is why is it rebooting? That detection fails if the hardware fails.

OXINARF, 2017-04-21:
The reason is pretty simple: some boards can't be recognized automatically.

As @magicrub said, there is no reason why the board is rebooting, it just gives a message over MAVLink. If it is indeed a Pixhawk clone it would be interesting to know why it isn't recognized though, it probably means it doesn't have the correct sensors.

I think this is a non-issue since you can set the board type manually if it isn't recognized.

floaledm, 2017-04-21:
Thanks for taking the time to educate me on this. Overriding the BRD_TYPE parameter "fixed" the issue so thanks for mentioning that.

Also, that clone seems to work fine (sensors respond and board does not reboot) with versions prior to 3.5-rc2 or when using the change found here: craftandtheory@0ac4b39

Is it a known/expected behavior for boards to reboot periodically if not all sensors are recognized? http://diydrones.com/forum/topics/pixhawk-error-check-brd-type-no-lsm303d-found
https://www.youtube.com/watch?v=JG6Vw0NbkhA
Is that a good thing to have it do that? On my end, this made it a lengthy process to get all the parameters as the board cycled through reboots...

Is it possible to not have these sensors properly recognized but yet have good data from them? I haven't tested that board in flight but mag, accel, and gyro values are reported, seem to accurately respond to movements of the board, are within expected ranges, and I don't have any bad/not healthy messages.

Anyway, is there a good reason to default to unrecognized board? If it's to prevent use of damaged boards, wouldn't the sensor fault be reported in another form anyway (e.g., sensor not healthy/not responding)?

OXINARF, 2017-04-22:
> Is it a known/expected behavior for boards to reboot periodically if not all sensors are recognized?

You posted two links and in neither of them I see a board reboot.

> Anyway, is there a good reason to default to unrecognized board?

I've already said why, some boards can't be recognized automatically, so a message needs to be displayed, we can't just assume it is a Pixhawk1.

>  If it's to prevent use of damaged boards, wouldn't the sensor fault be reported in another form anyway

No, if the sensor isn't detected there is no driver loaded for it so there is no unhealthy report. Also, one damaged IMU sensor shouldn't prevent anyone from using the board if that is their intent.

floaledm, 2017-04-22:
> You posted two links and in neither of them I see a board reboot.
My board and the ones in the links keep sending "Initializing APM..." then the board recognition error. I called it a reboot; I don't know if you'd call it a reinitialization or something else, but it keeps cycling.

> some boards can't be recognized automatically, so a message needs to be displayed, we can't just assume it is a Pixhawk1.
That's a valid point. I assume you could have a Ph2 with a faulty sensor, and there's no reason to detect it as a Ph1. I'm not sure the end result is different though than defaulting to unrecognized b/c:

> if the sensor isn't detected there is no driver loaded
and if there's no driver loaded, you can't communicate with the sensor, so it's even more indicative that there's a problem that unhealthy since the values are not reported at all.

Anyway, my intent with this issue was to bring up certain concerns to see if they should be addressed. I agree in the end that it's a non-issue since users can set the board type manually. On the other hand, maybe the documentation should be updated then because when I researched in the forum, wiki, and elsewhere about the "Unable to detect board type" error reported by my board, I couldn't find a clear answer about what that meant without looking at the code, and much less how to fix/bypass that...

OXINARF, 2017-04-23:
> My board and the ones in the links keep sending "Initializing APM..." then the board recognition error. I called it a reboot; I don't know if you'd call it a reinitialization or something else, but it keeps cycling.

It is cycling as it should, the fact it shows "Initializing APM..." is a just an unfortunate side effect.

> and if there's no driver loaded, you can't communicate with the sensor, so it's even more indicative that there's a problem that unhealthy since the values are not reported at all.

If it didn't load a driver because no device was there, why would it be unhealthy? You are assuming that when we try to load a driver a device *has* to be there, but it hasn't - boards use different sensors over time, that is why we check for several different ones on that method in px4_drivers (and that is where we check the devices that *have* to be present to be that board).

> On the other hand, maybe the documentation should be updated then because when I researched in the forum, wiki, and elsewhere about the "Unable to detect board type" error reported by my board, I [...]

Feel free to submit a PR improving documentation.

Anyway, this isn't a real issue, just another clone board that is doing something different but pretending to be a Pixhawk. If you know what sensors it has maybe we can fix it. Just making every undetected board be a PH1 isn't a good approach.

floaledm, 2017-04-23:
I agree with you. Thanks for the interesting conversation.

closed 2017-04-23 by floaledm