ArduPilot/ardupilot #6103: type_traits: No such file error when building with Qt on ubuntu 14.04

kyuhyong, 2017-04-24:
#### Issue details
_Please describe the problem, or desired feature_
OS: Ubuntu 14.04 LTS
arm-none-eabi toolchain 4.8.2 installed 
Qt Creator based 5.7.1 installed
Build setup done per instruction here http://ardupilot.org/dev/docs/building-px4-with-qtcreator.html
and build done in the first time.
But trying to build the next time, Compile output gives error saying 
 /home/dronedev/workspace/apmfirmware/libraries/AP_Common/missing/type_traits:4:28: fatal error: type_traits: No such file or directory
 #include_next <type_traits>
What should be done to resolve this issue? 

#### Version
_What version was the issue encountered with_
Latest origin-master
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
NA
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_
make[3]: Entering directory `/home/dronedev/workspace/apmfirmware/modules/PX4Firmware/Build/px4fmu-v2_APM.build/home/dronedev/workspace/apmfirmware'
CXX:     /home/dronedev/workspace/apmfirmware/ArduCopter/afs_copter.cpp
In file included from /home/dronedev/workspace/apmfirmware/libraries/AP_HAL/utility/functor.h:20:0,
                 from /home/dronedev/workspace/apmfirmware/libraries/AP_HAL/AP_HAL_Namespace.h:4,
                 from /home/dronedev/workspace/apmfirmware/libraries/AP_HAL/AP_HAL.h:5,
                 from /home/dronedev/workspace/apmfirmware/ArduCopter/Copter.h:28,
                 from /home/dronedev/workspace/apmfirmware/ArduCopter/afs_copter.cpp:5:
/home/dronedev/workspace/apmfirmware/libraries/AP_Common/missing/type_traits:4:28: fatal error: type_traits: No such file or directory
 #include_next <type_traits>
                            ^
compilation terminated.

khancyr, 2017-04-24:
Hello ! 
Just update the toolchain to 4.9 and it will solve the problem!

OXINARF, 2017-04-24:
@kyuhyong This list is for confirmed bugs and feature requests. For developer support please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

Like @khancyr said, you need to update your compiler to version 4.9.

closed 2017-04-24 by OXINARF

ranqingfa, 2017-07-31:
Got same problem and solved with gcc 5.4