ArduPilot/ardupilot #6105: Autotune not saving correct params on disarm

Gunaveh, 2017-04-24:
When performing an autotune in roll or pitch axis only, upon completion of autotune, pixracer does not save correct values. When testing the new values, after moving autotune switch from LOW to HIGH, quadcopter performs beautifully.  After landing in alt hold, waiting for motors to reduce to mot_spin_arm,  disarming, and disconnecting battery, the vehicle will not have the same values on next boot. If reconnected and flown immediately, values appear to be extremely low and result in a loss of control.

The param file including shows values from a successful autotune in yaw, user set values for pitch, and autotune values for roll upon reconnect.
3.5.0-rc4

210 size fpv quadcopter no gps 4s 2200kv tmotor f60 with bi-blade dalprop bullnose 5045

pixracer

#### Logs

https://drive.google.com/open?id=0B9egjsa9aEXWOU1sU3VWV1REOFk parameters

https://drive.google.com/open?id=0B9egjsa9aEXWRk1TaElnME1xeUE log

hdtechk, 2017-04-24:
You need to disarm still in AutoTune for new parameters to be saved

Gunaveh, 2017-04-24:
Yes. So once i turned on althold, i did not change it other than toggling the autotune switch for the duration of the flight.

Gunaveh, 2017-04-24:
I turned on stabilize on that one log just to show that it was using parameters other than the ones it saved with.

magicrub, 2017-04-24:
Are you flying in autotune mode?

rmackay9, 2017-04-24:
I've heard @lthall also complain that some parameters are not being updated after an autotune although I haven't been able to reproduce it yet.

Gunaveh, 2017-04-25:
Perhaps autotune is just unable to find good values for my particular setup.  I successfully ran autotune on another 210 frame that had gps but was using 3s batteries.  My issue with identical setup minus gps and 4s gives me repeatable post-autotune values of .01 for the rate p in pitch and roll, and values less than 1 for angle p. 

When I was flying in alt hold and using channel 7 to switch autotune, I noticed that a couple times it would perform nicely after switching to the new values. Then after landing and reconnecting I would have the same .01 p.  Today I flew using autotune as a 'normal' flight mode and switched between that and alt hold.  Upon completion of the autotune and switching between alt hold and autotune mode it was uncontrollable again.

Gunaveh, 2017-04-25:
https://drive.google.com/open?id=0B9egjsa9aEXWUW5RQXRlYUhRWEk todays last flight

Gunaveh, 2017-05-21:
This is still an issue. Autotune is unusable in pitch and roll axis

Gunaveh, 2017-05-21:
It appears that this has been an issue for about a year.  I found an old thread that describes the same type of failure I have been seeing. 

http://discuss.ardupilot.org/t/autotune-failed-pixracer-overpowered/9578/3

lthall, 2018-10-02:
It is saving the correct parameters but you are getting a terrible tune.

The question is why is the test flight part of the code not working.

Your gyro filter is way too low- I would suggest 80
INS_GYRO_FILTER,20

It looks like you increased these but I use 40 myself (half what I set the gyro filter to).
ATC_RAT_PIT_FILT,80
ATC_RAT_RLL_FILT,80

You are not using one shot or d-shot.
MOT_PWM_TYPE,0

You don't have fast logging turned on but I think your current tune is controlling better but also barely stable in rate.

So there are obvious reasons for autotune not working and I can see from your log that it is doing everything correctly from a tuning perspective.

I am going to change the name of this issue to Can't test autotune values with Aux Switch.

peterbarker, 2021-07-30:
We've changed the way this is done a few times since this issue was created.

closed 2021-07-30 by peterbarker