ArduPilot/ardupilot #6107: APMRover: add a way to dectect a broken steering

khancyr, 2017-04-24:
Idea from http://discuss.ardupilot.org/t/what-events-trigger-fs-crash-check/16697
Could potentialy be added to the crash check

rmackay9, 2021-07-14:
This idea came up again in this discussion during Rover-4.1.0-beta testing https://discuss.ardupilot.org/t/yuris-4-1-0-beta-update-gps-yaw-now-beta3/71100/118.

rmackay9, 2021-12-21:
This issue came up again in a discussion with a Partner on 21-Dec-2021

geofrancis, 2021-12-21:
@rmackay9  Would monitoring the difference between piddesired and pidachieved be enough to catch a steering failure?

rmackay9, 2021-12-21:
@geofrancis,

Basically yes I think that should work.  We actually have (in copter at least) a mavlink message that we publish which has the root-mean-squared difference between the two.  I don't think anything actually makes use of these calculations but this is probably the basis of what we should use for the control failsafe.

geofrancis, 2021-12-21:
@rmackay9 a boat could get into a situation where its some kind of environmental issue like the wind stopping it turning that could be interpreted as a steering failure. you would probably want some kind of weighting to it or at least a decent timeout so it doesn't give false alerts. There is the code in sail that checks if the boat is too far off course due to a failed tack and engages the motors that could be useful.

rmackay9, 2021-12-21:
@geofrancis,

Yes, similar to the FS_EKF_THRESH which allows setting the sensitivity of the failsafe..

geofrancis, 2021-12-21:
@rmackay9  yes, for something like a boat thats not going to have much around it could probably be relaxed a lot, compared to something like a lawnmower where you want much tighter tolerances.