ArduPilot/ardupilot #6109: Teraranger One I2C Conflict with 3DR Compass

tcheehow, 2017-04-24:
#### Issue details
Teraranger One and compass doesn't work together. Suspected I2C conflict as each can work independently if the other is removed from the I2C bus.

#### Version
Latest master

#### Platform
[  ] Copter

#### Airframe type
QUAD

#### Hardware type
Tested on Pixhawk and Pixhawk Mini

OXINARF, 2017-04-25:
@tcheehow I'm not sure what this report is about. If the I2C addresses are the same there is nothing ArduPilot can do about it, they need to be different.

rmackay9, 2017-04-25:
I think all we can do is update the wiki to tell people how to change the I2C address of the terrarange.  I'll take this one... maybe @tcheehow can give us details on how to modify the sensor's address.

Galaxy88, 2018-06-08:
To change the address of the TeraRanger One, you need to send the Change_Base_Address command 0xA2 followed by the new address to set, eg. send 0x42 if you want base address 0x42 (ignoring read/write bits). 

After sending the command, power cycle the device. It will join the I2C bus as a slave on the new address. 

[Source](https://www.terabee.com/wp-content/uploads/2017/09/TR_One_Manual_FW5.0.Version1_0_0.pdf)

peterbarker, 2021-07-30:
@Hwurzburg this could be a useful thing to document @Galaxy88 's fix in the Wiki somewhere.

In any case, not something for the ArduPilot repo.

closed 2021-07-30 by peterbarker