ArduPilot/ardupilot #6117: How to add a control controller

hiroyuki405, 2017-04-26:
Hello
I am currently designing a new controller
So, I'd like to improve the current Ardupilot controller, but please tell me which part should be changed.
Specifically, I would like to add a new controller after the PID controller at the following URL.
http://ardupilot.org/dev/docs/apmcopter-programming-attitude-control-2.html

Which function should I refer to?

magicrub, 2017-04-26:
Hi @hiroyuki405 !
You've posted a help/HowTo question in the developer bug tracking/featureRequest github issues area. instead, please repost in discuss.ardupilot.org.

closed 2017-04-26 by magicrub

rmackay9, 2017-04-26:
@hiroyuki405,
..But while we're all here... I think the new controller would go into the AC_AttitudeControl library somewhere.  https://github.com/ArduPilot/ardupilot/tree/master/libraries/AC_AttitudeControl.
Probably within the AC_AttitudeControl.h and .cpp files.