ArduPilot/ardupilot #6119: PX4: Booting without SD card

WickedShell, 2017-04-26:
#### Issue details
With the removal of the PX4 LED driver when we hang on startup because of no SD card present it's not clear to the user what happened. Because we moved the LED driver in tree, there isn't any way to turn it on at the NSH level anymore.

The proposed solution from @tridge is to allow us to always boot regardless of if the SD card is present, but have a parameter that says to quit and return to the NSH console if USB isn't present.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk

NickNunno, 2018-03-16:
I embarrassingly fell victim to this today... No sleep, working too long.. grabbed a random cube off the shelf and went down the rabbit hole trying to figure out why I couldn't flash/connect to this particular cube... Turns out no SD card. 

Not looking to point fingers away from my idiot move but honestly, it didn't even cross my mind as I surely thought there would be some kind of error, warning, "Hey dummy" message displayed notifying the user of a problem. No error sent me chasing my tail.. 

Point being, we don't seem to be failing gracefully here.

WickedShell, 2018-08-26:
We always boot, the issue is stale and should have been closed.

closed 2018-08-26 by WickedShell