ArduPilot/ardupilot #6122: “Bad LiDAR Health” even if "RNGFND_LANDIND = 1"  in arduplane firmware 3.8.0

ranqingfa, 2017-04-27:
“Bad LiDAR Health” even if "RNGFND_LANDIND = 1"  in arduplane firmware 3.8.0

 And when landing rangefinder really didnt work!

3.8.0 rc4

 Plane

Fixed Wing

pixhawk

Logs
https://pan.baidu.com/s/1pKIQRY7

Also why I can not upload a ZIP file?
[image]

ranqingfa, 2017-04-27:
I am using Maxbotic-I2C rangefinder, and in firmware 3.8.0 the RNGFND_TYPE = 2 for a Maxbotic-I2C rangefinder.

OXINARF, 2017-04-27:
This is a support question and those should be posted in the forum (http://discuss.ardupilot.org). I was going to let it stay because it had a log, but that link is goes to an invalid page.

Please post in the forum, together with the log - if you can't upload the log in the forum, then try to put in Dropbox or Google Drive or another hosting service and link it in your post.

closed 2017-04-27 by OXINARF

ranqingfa, 2017-04-27:
https://pan.baidu.com/s/1pKIQRY7
This is a baidu Drive.

ranqingfa, 2017-04-27:
Google Drive. Thanks a lot.
https://drive.google.com/file/d/0B6nj4A2GqXmWdFc2NTVRVWk4SjA/view?usp=sharing