ArduPilot/ardupilot #6123: Copter: Long running loops cause instability of flight and vehicle crash

hsteinhaus, 2017-04-27:
#### Issue details
Performance issues seem to cause single, very long running loops in APM:Copter. When that happens, the ESC signals of the vehicle do not get updates (0.45s in one case), leading to extreme instability of flight, random vehicle movements and to a vehicle crash in one case.

[image]

[image]

[image]

#### Version
ArduCopter 3.5-rc4

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
X4

#### Hardware type
PixRacer

#### Logs
https://github.com/DuraCopter/ardupilot/tree/glitch_longloop
https://github.com/DuraCopter/ardupilot/tree/crash_longloop

Note: these logs were recorded with a (very slightly) modified version of ArduCopter, however the is no indication so far that the instability is related to the changes. Due to the inherent risk of another crash, I am reluctant to try to reproduce this with master unless we have a good plan how to catch the bug.

Details discussed here: http://discuss.ardupilot.org/t/arducopter-3-5-rc4-long-loops/16832

rmackay9, 2017-04-27:
I've tagged this as an AC3.5 issue because I think that's the only place we've seen it.  It's probably the item that makes me most worry about releasing AC3.5 actually.

hsteinhaus, 2017-04-27:
I have started to put together some crash remainders and will see if I can get it airborne next week. There is only one problem: 9 out of 10 flights went fine (zero long loops), so there is a considerable risk of false negative test results. I think I will start with stick 3.5-rc4 and if I see glitches, I will directly switch to master.

peterbarker, 2017-04-28:
@hsteinhaus What are you using for telemetry on this beasty?

Can you describe your hardware setup, please?  Perhaps with a photo?

hsteinhaus, 2017-04-28:
RC and telemetry on this vehicle are indeed quite exotic: I am experimenting with the TBS Crossfire 868MHz (http://team-blacksheep.com/products/prod:crossfire_mrx ) radio link. The radio link promises to provide a transparent serial link that is aware of MAVLink. The system works quitewell, unfortunately it is a closed source blackbox.

At the time of the incidents there was a GCS connected (Tower). I cannot rule out that there was some MAVLink traffic involved, e.g. due to an initial parameter transfer.

General hardware setup:
 * 6s LiIon battery
 * AUAV ACSP4 powermodule
 * AUAV Pixracer (inverted mounting position, Roll180)
 * Drotek M8N XL GPS
 * SSD 1306 onboard I2C display connected to primary I2C (via GPS module)
 * TBS Crossfire Micro V2 receiver, connected to S.Bus in and Telemetry2
 * BLHeli_S based 4in1 ESC runing in OneShot125 mode
 * 330kv Motors, 17" propellers, 780mm quad, slightly above 3kg TOW

rmackay9, 2017-04-29:
Latest update from @tridge, @peterbarker is that we've found an issue in the handling of parameter refreshes.  It seems we can have a queue of requests from the GCS for individual parameters and we process these in the main thread and each takes 7ms (3x our regular loop length).  This is especially bad with high speed telemetry connections because the queue can get quite long.
We're working on a fix and this will be released with AC3.5-rc5.

peterbarker, 2017-04-29:
@hsteinhaud Any Mission Planner involved?

hsteinhaus, 2017-04-29:
No, only APM Planner (at home) and Tower (in the field).

peterbarker, 2017-05-01:
Is there any telemetry log available from Tower for the flight where the problem occurred?

A mitigation is going into MP, it is possible that Tower could do with one too.

hsteinhaus, 2017-05-01:
I think these files must be the corresponding telemetry logs:

http://steinhaus-home.de/cld8/index.php/s/rT70QBvdqpgHZCi (crash)
http://steinhaus-home.de/cld8/index.php/s/JHCkO6GcebS0zYb (gltiches only)

Regarding MP: I usually do not have any portable Windows equipment at hand on the field. Additionally, I have lost two vehicles due to the way Mission Planner is converting floating point numbers on systems with non-English locale settings, e.g. after MP was setting the PID gains to zero during flight (see https://github.com/ArduPilot/MissionPlanner/issues/156 and https://github.com/ArduPilot/MissionPlanner/issues/128). I am not sure if these issues have been fixed in a fundamental way in between (the comments of Michael suggested that they had not) and I am absolutely not keen on having crash number three due to that particular reason.

hsteinhaus, 2017-05-01:
I think I just reproduced the glitches right after triggering a parameter refresh in Tower:

[image]

Here is what I did this flight:
1. takeoff, short check of the copter in AltHold (Tower disconnected)
2. Loiter over a soft place (unfortunately in 20kt gusts, so the log might look a bit funny).
3. connect Tower (noticeable effects so far)
4. trigger a parameter refresh in Tower (repeated 3 times)

At the end of _each_ refresh, there were noticeable glitches, plus appropriate MaxT spikes in the log. The glich seems seems to happen _after_ tower has reached 100% for the first time, obviously during the re-reads of the parameters missed in the first approach.

This flight was still on my modified branch of 3.5-rc4, however on completely different hardware (original 3DR Pixhawk1 instead of PixRacer, no OLED display, no OneShot ESCs, different frame). Tomorrow I will cross-check the unmodified rc4 and the latest master.

hsteinhaus, 2017-05-02:
Results with stock Copter-3.5-rc4 (bd6acd96) - very nice glitches again at the end of Tower param refresh: 

[image]
[image]

Will take a flight with master in a few hours after the rain has passed.

hsteinhaus, 2017-05-02:
Just flown master. I did two separate flights, with lots of in-flight reconnects and parameter reloads. Each flight included some Loiter (in very gusty wind) plus some Auto and PosHold. To make it short: significant improvement, no noticeabe glitches (at least I did not recognize any in 15kt of wind) and ATT.ErrRP looks clean, too.

Here are a few pictures of the performance data:

**Flight 1**
[image]
[image]

**Flight 2**
[image]
[image]

As you can clearly see, PM.nlon has drastically improved, compared to 3.5-rc4. There are also no really long loops in the 100ms region. However, there is still one longer loop in flight 1, with 42ms. I guess, the changes form @tridge (60e4c0eb986450cc6bdc1aaa5ef5c914c98bf550..5c62e4f4c2a90670aec7ee69fdff1f83d27026ac) brought a huge improvement here - I hope they make it into an RC5 soon. However, the one long loop suggests that there might be a similar issue left open so far.

Gerneally speaking the impression of the flight performance was very pleasant, given the gusty weather and the poorly tuned copter. I did not notice any other unusual effects. I also gave Tower 4 beta a short try - worked very nice so far.

ThorstenDP, 2017-05-03:
I made several 20-40min min flights with a small pixracer/oneshot125 quad today running AC3.5rc5/3716630 and EKF3 (only) enabled. Flights were all fine. However, PM.NLon is still in the range of 110 and 180. But no Problems with MaxT and a slow telemetry with Tower.

If there is any interest here is a log (It was a little windy, the tuning is not yet perfect and there have been some fail saves...): https://www.dropbox.com/s/zw978ylu8dbgruj/2017-05-03%2011-20-23.bin?dl=0

hsteinhaus, 2017-05-03:
@ThorstenDP may be the NLon is related to the SD card? 

I did a short test flight (only some Loiter) to support/reject this theory and that is what happened with a stock Pixhawk 2GB card instead of my usual Samsung 16GB high speed card, with same amount of loggging configured as usual:

[image]

[image]

ThorstenDP, 2017-05-03:
@hsteinhaus 
Hi Holger, I bought this SD Card yesterday, because I had some troubles with with two other ones. It is an 8GB 60MB/s card. Before I was using a fast 16GB card. I changed the card because with the 16GB one logging stopped after about 9 to 12 min in several flights. This was not the case today. The other changes were: no RAW M8T logging today and EKF3 instead of EKF2.
In my previous tests EKF3 produced higher NLon values compared to EKF2. As soon as the weather permits I'll compare EKF2 and 3 again without changing other settings.

ThorstenDP, 2017-05-04:
@hsteinhaus 
Hi Holger, seems it was too late yesterday... So from your graph, if I interpret it correctly, NLon decreased to 0 when using the (slower?) 2GB  card!? I'll try some different cards as well as soon as it stopped raining. I am wondering if it is related to the size of the card or the formatting of the card.

hsteinhaus, 2017-05-04:
That is exactly what I am confused about. Only noticeable logical difference between cards: the number of sectors per cluster.

```
[root@atg release]# file -s /dev/mmcblk0p1 
/dev/mmcblk0p1: DOS/MBR boot sector, code offset 0x58+2, OEM-ID "mkfs.fat", sectors/cluster 8, Media descriptor 0xf8, sectors/track 63, heads 64, hidden sectors 135, sectors 3858489 (volumes > 32 MB) , FAT (32 bit), sectors/FAT 3761, reserved 0x1, serial number 0x2a25f76a, unlabeled
```

```
[root@atg release]# file -s /dev/mmcblk0p1 
/dev/mmcblk0p1: DOS/MBR boot sector, code offset 0x58+2, OEM-ID "mkfs.fat", sectors/cluster 16, Media descriptor 0xf8, sectors/track 16, heads 4, hidden sectors 2048, sectors 30701568 (volumes > 32 MB) , FAT (32 bit), sectors/FAT 14977, reserved 0x1, serial number 0xb15a37a7, label: "DC07       "
```

ThorstenDP, 2017-05-04:
Yes strange! I had (still have) some problems with camera trigger feedback. While desperately trying to solve it and previous formatting issues I played with the cluster size. Using a cluster size of 64kb I was unable to connect to mission planner. And it was not straight forward to reformat and install AC on it again. So maybe a lower cluster size does the trick. I'll try asap (raining all day so tomorrow).

BTW: how is cluster size related to sectors/cluster for FAT32? I just found this in some forum:
```
s/c size
 8  4K
16  8K
32 16K
```

ThorstenDP, 2017-05-05:
I made a short test with an 8GB card and a cluster size of 4kb. NLon is still in the same range (110 and 180). I'll try a 2GB card next.

[image]

peterbarker, 2017-05-05:
On Fri, 5 May 2017, Thorsten wrote:

> I made a short test with an 8GB card and a cluster size of 4kb. NLon is
> still in the same range (110 and 180). I'll try a 2GB card next.

We've noticed that the higher-bandwidth cards often have poorer 
performance for logging.

It's refered to as "stutter" - where the SD card goes away for a while to 
rearrange its internal storage a bit.  Those things are way too smart for 
us!

The smaller cards seem to be "dumber" and thus not prone to the stuttering 
as much.

We tend to use the Sandisk Unltra grey-band 8GB cards a lot.  Sadly, 
they're going out of production...

ThorstenDP, 2017-05-06:
@peterbarker  Thanks a lot for the information!

Some more questions:
- Are there any recommendations regarding formatting or is there no influence?
- What else could be done to mitigate the influence on Ardupilot performance? I have not seen a big difference when switching off GPS RAW data logging.
- What are a critical NLon and MaxT thresholds?

rmackay9, 2017-05-08:
I think we've resolved the issue that led to the creation of this issue - i.e. the vehicle crashing because of the long loops.  So if there are no objections I'll close this item.  It's actually possible to keep discussing other things here but I think it's fair to close the issue.

closed 2017-05-08 by rmackay9