ArduPilot/ardupilot #6127: PWM RC_Output no longer refresh and leads to crash.

cyrilcalvez, 2017-04-27:
In a 28 minute flight with a S900 DJI platform with E1200 pro arms powered by a Droppix with Ardupilot 3.4.5, the RC_out (RCOU.C1 to RCOU.C6) from the pixhawk no longer refreshed and leads to a freefall and crash. At approximately 70 meters high it began to roll and freefall.
Here is the log of the RC_out:
[image]

All the PM parameters logs seem to be normal (no long running loop). IMU, gyro, baro are ok.
EKF2 and NKF logs are ok.

I have seen that Mr Steinhaus have encountered the same problem with a pixracer here: [https://github.com/ArduPilot/ardupilot/issues/6123](url) but we don't see any long loop on our side:
[image]

Do any of you have any idea where does this come from?

Thanks,

Cyril

#### Version
Ardupilot 3.4.5

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
S900 hexacopter airframe with E1200pro

#### Hardware type
Droppix (Pixhawk version from Drotek) with Ardupilot 3.4.5

khancyr, 2017-04-27:
hello,
could you like the log file so that we can look at it ?

cyrilcalvez, 2017-04-27:
I can provide log but i don't know if there is an easy mean to remove gps data location from it?

rmackay9, 2017-04-27:
I think Tridge and I saw this posted somewhere in the discuss forums as well (although I can't find the posting anymore).  We think the I/o board (the 2ndary CPU on the board responsible for input and output) must have crashed.

I think Tridge's last question on the discuss forum was asking what transmitter/receiver is being used.

I think we found the usb was plugged in as well which isn't good although I believe the reporter said that it's been that way for a couple of years without any problem.

cyrilcalvez, 2017-04-27:
Thanks for you answer.
I have removed lng and lat data from log file and kept only the last tenth of seconds to keep the size small.

[27_04_2017.txt](https://github.com/ArduPilot/ardupilot/files/961544/27_04_2017.txt)

Nothing is connected on the usb port of the Droppix.
Onboard voltage is constant at 4.7V.
We are using X8R receiver from FRSKY.
Thank for the info, i will continue to investigate on the side of the second I/O CPU.
After the freeze of RC_OUT (at 367.816 in timestamp) we cannot see if the RC_IN is also freezed, but it doesn't change for these last 2 seconds.

cyrilcalvez, 2017-04-27:
I have just seen that the Vservo doesn't get updated from the 368.816 point:
[image]

OXINARF, 2017-04-27:
@rmackay9 I believe you are thinking of #5859. Indeed this looks similar.

@cyrilcalvez Thanks for looking into it. The lack of updates to Vservo also points to lack of communication with the IO CPU.

cyrilcalvez, 2017-04-27:
Thanks for the input OXINARF, it seems that we have encountered the same probleme as issue #5859.
We are not using Lidar, compass motor compensation is off.
Gimbal was disconnected, only the X8R Frsky receiver and the 6 ESC were connected to the servo rail (DJI opto esc without BEC).
The servo rail was only powered by the +5Vcc of the pixhawk (Power input with current and voltage adc).

cyrilcalvez, 2017-04-27:
X8R draw current from the Vcc side and not servo rail in the Dropix.
We measured current draw at 60mA for the X8R and Pixhawk draws 190mA. Results are the same on an identical functional unit.
We are going to see if the unit is working again, we did not test it before as the crash seems to have broken all the parts (plastic cover, wire, etc)

cyrilcalvez, 2017-04-27:
The Dropix crashed unit is working, receiver SBUS data is transfered to pwm AUX output and main output. I assume that the STM32F100 is ok.

cyrilcalvez, 2017-04-30:
We are actively using ardupilot stack for all of our UAVs but this issue will change our mind about it there is no clue to solve this serious bug of the system. As the hardware is ok and doesn't seems to have failed in flight, what can i do to find the cause of the crash or/and of the 2nd IO/CPU?

OXINARF, 2017-05-02:
@cyrilcalvez We talked about this in dev call today. This is very hard for us to debug because this happens only to two users (assuming it is the same in both issues) and not to any of us - the fact that there is no logging or anything post-mortem from the I/O CPU also doesn't help as we only know the link was lost. What we would like to ask you is to try and reliably reproduce it **on the bench**. If you can do that and give us a list of steps to make it happen, then we'll ask someone on our team to reproduce it in our hardware - hopefully we don't need to get to the point where we need your hardware to reproduce it.

I'd like to point out that this is on our list and a big worry to us, but our hands are kind of hands tied without being able to reproduce.

cyrilcalvez, 2017-05-02:
Thanks for your reply. We will try to reproduce the bug on a bench. We were using pixhawk (3DR version) without any problem for more than a week when used in a bench, we will try to replicate with a Dropix.
I have seen that we are using the same Dropix unit as the other issue #5859 . The Dropix version is V2.1.
I will send a mail to Drotek to see of other users have the same problem.

cyrilcalvez, 2017-05-02:
I have written a post on Drotek support boards to see if there is something on there side.
[https://forum.drotek.com/t/crash-of-the-io-cpu/1051](url)

If the IO_CPU resets, because of EMC for example, does it get back to a normal operation on its own or does it have to be initialized by the master CPU? And how long does it take to get back to normal operation?
Thank you!
Cyril

cyrilcalvez, 2017-05-03:
On the Dropix board there is no connection between the Vbat pin and 3V3 VDD as in the PX4IOV1.3 schematics. The Vbat pin seems to be floating. Can this be the cause of an unexpected reset of the MCU in flight?

ClementChristomanos, 2017-05-17:
Hi @cyrilcalvez ! By any chance did you get some more info regarding that crash ? Could you tell us a bit more about the environment you were flying in (via PM if necessary) ?

khancyr, 2017-05-17:
@ClementChristomanos @cyrilcalvez, were you flying with a power tethering device ?

cyrilcalvez, 2017-05-18:
Hello!
Yes, we were flying with our power tethering device. The place was a desert, with only a generator to get electricity. There was no high voltage line or other equipment that can produce EMI.
We have flown hundreds of hours with different drones, payload and environment without any problem. It's the only time it happened and we didn't succeed to reproduce it on the bench after that.
Is the voltage measured at 4.7V vor Vservo and pixhawk supply instead of standard 5V can cause some problem?

ClementChristomanos, 2017-05-18:
Hi Cyril,
Same for us, we weren't thethered. The place was desert too. 
Same for us, this particular machine flew for hundreds of hours. 
I doubt the voltage would be an issue (but I remember reading about an undervoltage that could cause brownouts), our values show Vcc at 4,98V and VServo at 5,15V.
The day it happened we actually two machines, one small test machine (F550) and a larger one. The F550 crashed (but at low alt, so no damage). We did not find anything disturbing in the logs except the RC Outputs, we flew again and it flew fine for 15 mins and crashed again. We suspected a power issue on the small drone and decided to flew the larger one. We flew at a different place for the whole afternoon without any issue. Coming back to the first place, it crashed after 10 mins flight.
Mistery... I would not be surprised if our Dropix's came from the same production batch.

wangxuanjie, 2019-04-04:
[image]
I have same problem too, and the crystal oscillator used in io mcu is faulted，it's hardware problem.

ClementChristomanos, 2019-04-04:
@wangxuanjie what hardware are you using more precisely ? Thanks for that feedback !

wangxuanjie, 2019-04-04:
I use PX4FMUv2.4.5 schematics, and re layout

IamPete1, 2021-08-30:
Any updates? there have been so many changes since the original report I'm afraid there is not much we can still learn from it.

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1