ArduPilot/ardupilot #6137: Arducopter 3.4.4 strange Pixhawk 2.1 Baro 1 behaviour, unable to disarm via CSC.

JacksonUAS, 2017-04-28:
#### Issue details
Multiple short flights carried out, found throttle stick to altitude hold performance inconsistent in Loiter and Alt hold modes.

Inability to disarm via sticks, had to disarm aircraft via mavlink.

Logs show Baro 1 reported altitude incorrect across multiple flights. Baro 2 looks normal. 

#### Version
Arducopter 3.4.4 Quad. 

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk 2.1 initial release. 

#### Logs
Log #41 experienced inability to disarm via CSC. 
https://www.dropbox.com/sh/dzleawd6qao4psd/AABZLuwc-c3HuzGB9dBCnXCJa?dl=0

magicrub, 2017-04-28:
If the baros were drastically different then that hints at hardware failure. Did the baros agree later during other flights?

JacksonUAS, 2017-04-28:
Nope, it appears to be a hardware failure, and at the same time Arducopter is failing to deal with it. Ive changed the primary baro to the 2nd one, which looks to be working. Ill do some reading into how the baros are fused into the EKF, but it doesnt look like it uses both of them, only the primary and when it spits out bad data that bad data is used?

JacksonUAS, 2017-04-28:
Looking at the EK2 documentation, and the code (and am not a coder!) it appears that only one barometer is fed to both EKF's, decided on boot if primary_baro is healthy (healthy being that it reports in 0.5s). Is this correct? Legacy code to support earlier hardware with one baro?

OXINARF, 2017-04-28:
@JacksonUAS This is a support issue and those should be posted in the forum (http://discuss.ardupilot.org).

EKF always uses the primary baro and the primary baro is the first healthy one (or the one chosen by the parameter, as long as it is healthy).

Regarding not being able to disarm, if you weren't in a manual mode, that's expected if the landing wasn't detected.

If you need more answers please post in the forum.

closed 2017-04-28 by OXINARF