ArduPilot/ardupilot #6145: Plane Fail Safe - Glide towards the wind

fredrikfalkman, 2017-04-28:
#### Issue details

_Background_

If you encounter a GCS comms loss when flying BVLOS, a RTL fail safe might not be desirable since it could mean flying in the blind for an extended time and distance.

For a plane big enough for a parachute that is currently a fail safe option.

For the project I'm working on - BVLOS flying for maritime SAR - we have concluded together with the local CAA that being very small and light will be a prerequisite for flying BVLOS in unsegregated airspace. Because of this we hope to avoid carrying a parachute.

The final current fail safe option i "Glide", setting flight mode to FBWA (with no throttle) which I suppose means that the plane will glide in the direction it happened to head in when the fail safe kicked in. If that direction happens to be with the wind the plane risks ending up a long way from the fail safe point. It would also risk landing/crashing with an unnecessarily high speed.

_Suggestion_

- Replace the current "Glide" fail safe with "Glide towards the wind"
- Additionally, it would be even better if the plane would first circle (or loiter, if health GPS) to a predefined (lower) altitude before gliding towards the wind. 
- In the end I suppose it would be nice if the plane attempts an auto landing based on altitude. 

Looking forward to the thoughts of this awesome community!

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[-] Plane
[  ] Rover

geofrancis, 2017-04-28:
I don't understand your need for it, arduplane already has a 2 stage failsafe, so it will circle until it enters long failsafe and begin to RTL. 

RTL works very well even in high winds. The only condition is that you must have enough power to accelerate and regain airspeed as you make a downwind turn. If you have the throttle slew rate too low or have inadequate power you have a serious risk of stalling.

Once the plane is flying downwind to the RTL point it will turn and begin to circle overhead. I have tested this in 50mph+ winds flying over mountain and it always worked as long as you have enough power for the downwind part and have set a minimum ground speed or you risk flying backwards. If wind speed is higher than your minimum airspeed and you don't set minimum ground speed it won't try and push through a headwind it will only give it enough throttle to hold altitude. 

Your suggestion of flying into the wind would only take you further away and it would then crash land wherever you lost control compared to RTL that would bring it back the fastest most direct route. It would then hold at its RTL Altitude circling above its launch site or preset rally point until you regained control.

But your idea would be useful in other failsafe conditions like loss of gps in auto. Currently it's only failsafe is circle. In high winds it can get far away fast, and could land very fast if its flying down wind. if it pointed into the wind it could cover less distance if the wind was fast enough relative to the minimum airspeed of the aircraft and it would land at its slowest speed.

fredrikfalkman, 2017-05-03:
Hej! Thanks for the reply!

Yes, RTL works fine if that's what you want the plane to do. But if you are flying BVLOS, say 20 km away, or say with the pilot somewhere else completely, in an airspace which permits it, with sufficient traffic awareness by means of ground based transponder reception, then you wouldn't want the plane to fly for an extended time without control.

The first stage failsafe is fine circling for a limited time in hope of reestablishing connection with the GCS. The time you can permit for that would depend on the separation from other traffic.

But if you go to a stage two failsafe under these conditions, then you would like the plane to land as soon and as gracefully as possible; Soon to avoid mid air collisions; Gracefully to minimise the risk of injury to people and property on the ground, as well as to save the plane.

So circling to a pre-set (lower) altitude and then landing into the wind I think would be the desired behaviour.

geofrancis, 2017-05-15:
The issue i see is that if your flying and hit the end of your range then decending is the last thing you want to do as you will almost certanly never get your signal back and your crashing wherever you lost signal. but if it does rtl you would get control back within a minute or so as soon as it turns around and flies back into range. 

if it was radio failure your worried about then use 2 radios, or some other independant telemetry like like 4g.

fredrikfalkman, 2017-05-15:
I see your point. Maybe you could have an option to backtrack for a choosable time before automatic landing into the wind. That way you have a chance of reestablishing comms, but still for no longer than the marginal you have to other traffic.

4g is primary radio. Secondary radio not always practical.

Fredrik Falkman

[cid:E39549BE-AAF0-468E-BB80-C5C7DDE5ED50]

Swedish Sea Rescue Society
Talattagatan 24; 426 76 Västra Frölunda
ssrs.se<http://ssrs.se/>  first-rescue.org<http://first-rescue.org/>  surtsey.org<http://surtsey.org/>

LinkedIn  se.linkedin.com/in/fredrikfalkman<http://se.linkedin.com/in/fredrikfalkman>
Twitter  @fredrikfalkman
Skype  fredrikfalkman
Cell  +46-705-280058

Se om du passar för världens viktigaste extrajobb – frivillig sjöräddare! Läs mer och ansök på vår hemsida. Och dela gärna filmen så att fler får chansen. Se filmen här!<https://www.youtube.com/watch?v=55Z6UZZUEdI&feature=youtu.be>

geofrancis, 2017-05-16:
there are rally points that can be set so if you enter RTL it will go to the closest rally point and loiter, you could set these to places and altitudes you know you get 4G so that it wont try and fly all the way home but just to the closest rally point to reaquire signal.

you could use a second 4g modem on a different network to give redundancy. that way it should mitigate the risk of loosing a single connection.

fredrikfalkman, 2017-05-17:
The thing is that I'm flying under RPAS flight rules, which means that autonomy must be kept to a minimum. Flying to a rally point would mean autonomous flight without control, with no way to fulfil the obligation to avoid other traffic. 

You are of course right that avoiding comms loss in the first place is paramount. We are exploring roaming SIM cards to get all cell networks as one way to do that. Monitoring cell reception to preempt comms loss might be an other. 

But if we still loose comms, landing on the spot after a short period of circling is what our CAA demands (rightly, in my opinion). Landing into the wind after circling to a lower altitude seems to me to be the safest way to do that.

geofrancis, 2017-05-17:
does your aircraft have an adsb transponder? then it could avoid and be seen by other aircraft http://ardupilot.org/copter/docs/common-ads-b-receiver.html

fredrikfalkman, 2017-05-17:
I Europe to here is still no mandate for ads-b on aircraft below 5700 kg tow. But some lower airspace users have TCAS, so that helps. Still, awoiding collision is a two party responsibility.

geofrancis, 2017-05-17:
the rally points would still be your best bet, if you know your route in advance you just set the rally points along the route so it wont deviate from the path you have defined. if it looses signal it will go to the closest rally point and loiter at your RTL altitude. 

if your flying over water you could set your rtl altitude to something very low like 40m then it will circle down over the rally point until it reaches 40m where it will loiter at minimum throttle untill the battery runs out, then it will glide down to a landing and with a lidar you would have very accuate measurement of where the surface is, 

by making it loiter low it will be clear of any other aircraft, and your giving it the maximum amount of time to regain contact.

failing that, there is a flight termination system that you can enable, it will bring it aircraft down but not in a controlled manner. its more of a self destruct mode.

http://ardupilot.org/plane/docs/advanced-failsafe-configuration.html

fredrikfalkman, 2017-05-17:
Yes, I know about advanced failsafes. 

From a regulatory standpoint, one meter above the ground is still flying, and while I concur with you that flying low means less risk of colliding with someone, it's not nonexistent. And as I mentioned, the regulator will not except that level of autonomy.

Today, setting the long failsafe to "Glide", is my only viable option. If it could do that into the wind it would be so much better!

geofrancis, 2017-05-17:
it sounds like the FTS would do it but you want the aircraft to survive the landing, so you want it to terminate but still maintain attitude control rather than putting servos to maximum positions. 

as far as i know the FTS can be programmed with its own "mission" if its acitivated so i assume land mode can be a part of that.

fredrikfalkman, 2017-05-17:
Attitude control + into the wind. Huge difference landing into 10-15 m/s winds as opposed to having the same wind speed as tailwind. Landing speed ≈0 vs ≈25-30 m/s.

geofrancis, 2017-05-17:
soits not so much a glide mode you need but just for the auto landing sequence to use the wind direction to set its final landing heading.

peterbarker, 2017-05-18:
On Wed, 17 May 2017, geofrancis wrote:

> soits not so much a glide mode you need but just for the auto landing
> sequence to use the wind direction to set its final landing heading.

Could you just enable the deep-stall landing code to get this effect?

WickedShell, 2017-05-18:
@peterbarker I suspect not. You need a subset of fixed wing types to try it safely. Most vehicles won't have sufficient elevator deflection to get into a stable one. (I heard @gmorph likes trying to destroy his fixed wing platforms though...). Also given the desire in this to minimize autonomy, it's a bit weird because deepstall is probably the most autonomous the aircraft ever gets when performing a mission. (Self selects and fly's paths more complicated then just a line based on the current external environmental factors). ATM all landing points also need to be programmed into the mission, which I think won't work to well for this use case. (Although I am planning a pass to allow deepstall/landing to be a proper emergency/failsafe response)

fredrikfalkman, 2017-06-07:
Is there anything I can do to make this happen? Lacking the programming chops to do it myself, I might be able to find a little bit of funding, provided that it would be - as I believe - a rather limited undertaking.

geofrancis, 2017-06-08:
i still dont see the need for this, your flying a small foam plane out to sea, if it goes down does it matter how it lands if your never getting it back?

fredrikfalkman, 2017-06-08:
I will be flying routinely in uncontrolled, unsegregated airspace. There will be manned aircraft around, such as the SAR helicopter. They will not be happy if loose control and one of my planes venture back on its own on RTL, so thats not an option.

Glide as a failsafe is OK from an airspace perspective, but there may also be persons on islands or on the water. Even if the likelihood of hitting someone is slim, I'd rather do it slowly then with a lot of speed. 

Let's say that you have a FBW airspeed of 12 m/s and flying in 10 m/s winds. That gives you a ground speed of 2 m/s flying into the wind and 22 m/s with a tail wind. Let's further assume that you have TOW of 2 kg. That would mean an impact energy of 4 Joules flying into the wind 484 Joules with a tail wind. 

Finaly, standard landings will be onto the water (because there will usually be a rescue boat on scene eventually which can retrieve the plane. I believe there would be a greater risk of loosing or destroying the plane if it glides a longer distance and touches down at high speed than doing the opposite into the wind.

WickedShell, 2017-06-08:
I don't understand your logic, as RTL can go to planned rally points which is a predefined location. From a manned aviation perspective the number one thing is being able to predict and tell them what the aircraft will do. Flying to a rally point is far superior then turning and gliding randomly into the wind as it makes your loss of link scenario predictable.

If you want to really minimize the energy on landing then you need something more reliable then just the wind robbing you of forward airspeed. At that point you need a parachute, planned landing (w/ reverse thrust or deepstall). I get that it's less energy, but it isn't enough to say "the system is now safe" unless your only goal is to perpetually be at or below the airspeed. And given that this makes the aircrafts behavior far harder to predict and interact with I have to say I don't think it's a worthwhile tradeoff.

From my point of view I don't get it as feature and it's not on my radar to implement, but you might be able to find someone else to implement it :)

fredrikfalkman, 2017-06-08:
Maybe a primer on my SAR use case might help clarify things a bit: http://www.surtsey.org/projects/cloud-control-station/

The missions will have minimal time for pre-planning and thus creating rally point will be impractical.

Flying below 120 m would mean a limited time in a glide failsafe. Minimising the time in the air when out of control while also minimising impact energy is what I'm after. Adding a parachute means added weight, which will lead to bigger batteries which will lead to added weight...

We will not get to a "totally safe" system any time soon, but for this use case I believe that (circling down, followed by) gliding into the wind would be a good compromise. A proper landing with deep stall or reverse thrust would be nice, but geting a good altitude reference over water again adds weight so might not be worth it.

magicrub, 2017-06-08:
end the auto mission with:
DO_LAND_START
LOITER_TO_ALT (with alt = 0)

when an RTL kicks in, it will simply loiter down right there

magicrub, 2017-06-08:
It would be nice to add additional failsafe behaviors. Landing into the
wind is not something I would personally want to do, but you might. Pull
requests welcome!

geofrancis, 2017-06-09:
fredrikfalkman when i say terminate, i mean use the flight termination system to crash the plane if RTL does not get you back in range after 2 minutes.

magicrub, 2017-06-09:
like this! https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp#L350

fredrikfalkman, 2017-06-09:
As I have tried to point out above, I don't wish to crash the plane. I want the opposite; an uncontrolled landing with as little impact force as possible, both for the sake of the plane and for the sake of persons and property that might be in the way.

geofrancis, 2017-06-09:
How heavy is your plane?

fredrikfalkman, 2017-06-09:
1.5 - 2 kg, in it's current implementation. Striving to go lighter.

geofrancis, 2017-06-09:
i have stalled foam planes that weight and crashed them into long grass, it will usually be undamaged so its safe to say it would probably survive anything but a high speed dive into water.

fredrikfalkman, 2017-06-09:
Yes, it shouldn't be a problem if the speed isn't too high. Into the wind!

fredrikfalkman, 2017-09-06:
Hej community! If someone would work on this as a consultancy gig I might be able to find a little bit of funding. Not sure how that kind of thing works around here - please educate me!

tridge, 2017-11-14:
just saw this. I think adding a FLY_INTO_WIND waypoint type would make sense, perhaps taking a glide slope as a parameter, it could be combined with LOITER_TO_ALT to do what @fredrikfalkman wants

fredrikfalkman, 2017-11-24:
That sounds like a very good solution!

agising, 2024-06-28:
[PR #27413](https://github.com/ArduPilot/ardupilot/pull/27413)

Ryanf55, 2025-01-04:
This is now merged. yay. Open a new issue if it doesn't meet your needs

closed 2025-01-04 by Ryanf55