ArduPilot/ardupilot #6161: ground mag anomaly with PX4Flow

Ahrovan, 2017-05-01:
When using (Copter-3.4.6) without PX4Flow (disabled). flying good without any problem. but when Enable PX4Flow and take-off on stabilize/loiter mode . . see error and crash. 

EKF2 IMU1 ground mag anomaly, yaw re-aligned
EKF2 IMU0 ground mag anomaly, yaw re-aligned
EKF2 IMU1 is using optical flow
EKF2 IMU0 is using optical flow
EKF primary changed:0

EKF primary changed:1

PreArm: check range finder
PX4v2 00320038 33355108 33373538
Frame: QUAD
PX4: de6b667d NuttX: 8c965992
APM:Copter V3.4.6 (e707341b)

magicrub, 2017-05-01:
... and crash? Please post logs.

OXINARF, 2017-05-01:
This is a support question and those should be posted in the forum (http://discuss.ardupilot.org). We'll also need to provide more info there, together with a log.

closed 2017-05-01 by OXINARF