ArduPilot/ardupilot #6163: Q_VFWD_GAIN altitude above ground thresold

raptortrias, 2017-05-01:
#### Issue details
Use altitude above ground [ideally allow for barometric/GPS as well as sonar] to disable forward motor when aircraft takes off in position VTOL mode but the aircraft does not provide sufficient clearance for the forward motor to be used when on the ground. A couple of meters above ground would probably be sufficient. Of course if sonar is used it would be much safer especially during landings.

#### Version
N/A

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover

#### Airframe type
Fixed wing wing, VTOL

#### Hardware type
N/A

#### Logs
NA

magicrub, 2017-05-01:
It already does this. Is there a problem with the current system? 

From 0-1m the fwd throttle is 0, then from 1-5m or so it's scales up to full demand.

raptortrias, 2017-05-01:
Thanks,
 Couldn't find if there was a parameter to adjust this and I wrongly assumed it wasn't there.

 Thanks

OXINARF, 2017-05-01:
So can this be closed?

raptortrias, 2017-05-01:
Yes thanks

closed 2017-05-01 by OXINARF

magicrub, 2017-05-01:
It's not exactly as I explained, but close. Q_VFWD_ALT is the param you're looking for and it controls [this code here](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/quadplane.cpp#L2214).

It will limit the forward throttle during VOTL mode to zero when below that altitude. It is always scaled over the first 2 meters for smooth transitions above that. The default alt is 0, so by default at 1m you'll be limited to 50% demanded throttle and 0.5m would mean 25% throttle is still allowed. If you want the fwd motor to always be 0% when below, lets say, 3m then set Q_VFWD_ALT =3. In this example alt <= 3m the throttle is 0%. alt = 4m is 50% and 5m is 100% throttle (smoothly interpolated between those values).