ArduPilot/ardupilot #6164: Duplicated parameters in the documentation

amilcarlucas, 2017-05-01:
#### Issue details
The following parameters get generated twice in the ardupilot\Tools\autotest\param_metadata\apm.pdef.xml Parameters.html Parameters.wiki Parameters.rst files
ATC_RAT_RLL_P
ATC_RAT_RLL_I
ATC_RAT_RLL_IMAX
ATC_RAT_RLL_D
ATC_RAT_RLL_FF
ATC_RAT_RLL_FILT
ATC_RAT_PIT_P
ATC_RAT_PIT_I
ATC_RAT_PIT_IMAX
ATC_RAT_PIT_D
ATC_RAT_PIT_FF
ATC_RAT_PIT_FILT
ATC_RAT_YAW_P
ATC_RAT_YAW_I
ATC_RAT_YAW_IMAX
ATC_RAT_YAW_D
ATC_RAT_YAW_FF
ATC_RAT_YAW_FILT
H_SV1_POS
H_SV2_POS
H_SV3_POS
H_RSC_PWM_MIN
H_RSC_PWM_MAX
H_RSC_PWM_REV

#### Version
master HEAD

#### Platform
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Not relevant

#### Hardware type
Not relevant

#### Logs
Not relevant

meeththakkar, 2017-07-08:
I would like to solve this issue.

meeththakkar, 2017-07-08:
is this issue about same names used at two different places ?

auturgy, 2017-07-08:
No, it's because heli and multi are both within copter. The script generating the param list sees both, so generates that part of the param list twice, even though  only one will be active on a flight controller.

meeththakkar, 2017-07-08:
can this be happening because script doesn't know what is getting compiled  ( copter or heli) ?

magicrub, 2017-07-08:
The script doesn't compile the source, it just scans the source and collects the data.

Does anyone have a link to the script that they can share with us?

WickedShell, 2017-07-08:
https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_parameters.sh

auturgy, 2017-07-10:
I'd suggest that the fix here isn't to remove "duplication" but to clearly publish separate Heli and Multi lists. Thoughts?

rmackay9, 2017-07-11:
@auturgy, yes, that would be nice.  I worry about the downstream impacts for GCSs who would need to know where to get the parameter list when heli firmware is used but I suspect they could handle the change.

IamPete1, 2021-08-16:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/14763

closed 2021-08-16 by IamPete1