ArduPilot/ardupilot #6180: The same movement GCS-alt = 52m in tlog, but Baro-alt = 79m in log? Why?

ranqingfa, 2017-05-02:
The same movement GCS-alt = 52m in tlog, but Baro-alt = 79m in log.
You can see in the log file at the time about 10:04:45.
[image]
[image]

v3.7.1
Plane
Fixed Wing
Pixhawk

Logs(tlog and log)

https://drive.google.com/drive/folders/0B6nj4A2GqXmWeDFrZldwOTZrT1U?usp=sharing

OXINARF, 2017-05-02:
@ranqingfa This is the fourth time you are being warned about not posting here but in the forum - next time we'll need to take further measures.

You are seeing different altitude references, the one in tlog is altitude above home and the dataflash shows baro alt. There were various changes around this for the future Plane 3.8 release.

closed 2017-05-02 by OXINARF

ranqingfa, 2017-05-03:
@OXINARF  Very very sorry for this. First actually I dont know if this is a issue question. Second, there is usually no ansower in the forum.  At last, want to ask the alt above home is not equal to baro alt?

rmackay9, 2017-05-03:
These are definitely questions not issues.  "Issues" are confirmed bugs or enhancement requests.
It's tough to get answers some times on the forums because it's mostly volunteers - we rely on people (like you) to help each other as much as possible.

The answer to your question is that we have various altitudes based on various origins.  So there's the "alt above sea level", the "alt above home", "alt above terrain" and "alt above ekf origin".  the baro alt is most likely an "alt above ekf origin" which is close to home but not exactly the same, especially if home has been moved somehow.

As Francisco says, please don't keep posting on the issues list - really sorry, we appreciate you using ArduPilot but remember, we need this list to track our real bugs and enhancements.  Questions go in the forums.

ranqingfa, 2017-05-03:
@rmackay9 Thank you very much for the answer. When the plane flys, it uses alt above home?

ranqingfa, 2017-05-03:
And in the log there is no alt above home, I only saw the baro alt and GPS alt, ahrs alt.

rmackay9, 2017-05-03:
Let's take the discussion over to the forum..

ranqingfa, 2017-05-04:
@rmackay9 @OXINARF  Now I think this is a bug. See same question from the forum.  Same problem as mine.
http://discuss.ardupilot.org/t/constantly-20-30m-under-target-altutitude-and-not-correcting-auto-mode-mission/17059

ranqingfa, 2017-05-04:
From the rangefinder alt(100m), we know the plane flys at the right alt(100m), but the GCS alt is not right(90m),  usually lower >10meter, and the plane does nothing. Rusult in bad landing accuracy!