ArduPilot/ardupilot #6183: copter 3.5 pi zero not launching

awright424, 2017-05-03:
#### Issue details
I am using a diy pxfmini and I have been trying to figure out my esc issues with a dys f20a esc. I only have one motor spinning. I have went through and tried compiling the latest arducopter 3.5 and when i attempt to run on the pi zero it just crashes without any hint as to why. nothing in dmesg alluding to why as well.

#### Version
copter 3.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
x quad

#### Hardware type
pxfmini pi zero

#### Logs
nothing relevant yet.

awright424, 2017-05-03:
this may not be an arducopter issue. I was able to build this prior and tried with the 3.4 branch as well. but the executable created doesnt seem work

OXINARF, 2017-05-03:
@awright424 You need to post this in the forum (http://discuss.ardupilot.org), which is the appropriate location for user support.

I'm not sure why you are trying to compile, but we do builds for PXFmini:
- 3.5 (all frames except heli): http://firmware.ardupilot.org/Copter/beta/pxfmini/
- 3.4 (quad frame): http://firmware.ardupilot.org/Copter/stable/pxfmini-quad/

closed 2017-05-03 by OXINARF

awright424, 2017-05-03:
I need to build it because the outputs I use on the pcs9685 are different
and mpu9250 orientation is different as well.