ArduPilot/ardupilot #6184: Copter 3.4.4 cannot be disarmed

weety, 2017-05-03:
#### Issue details
Copter-3.4.4 version, the UAV can not be disarmed. If the UAV on the ground in the case of stationary power on state, LED has been flashing yellow light, can not be disarmed, even wait for a longer time, and then through the data transmission equipment state, GPS can search to 12 stars. If at this time will be placed on the right side of the UAV, the status of LED lights flashing green immediately after the normal flight can be disarmed.

#### Version
Copter-3.4.4

#### Platform
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover

#### Airframe type
F450 QuadCopter

#### Hardware type
pixhawk

#### Logs

weety, 2017-05-03:
Logs：
Log File C:\Users\luohui\Documents\Mission Planner\logs\QUADROTOR\1\2017-05-03 20-15-37.log
Size (kb) 9111.6650390625
No of lines 109077
Duration 0:05:47
Vehicletype ArduCopter
Firmware Version V3.4.4
Firmware Hash 8a196281
Hardware Type 
Free Mem 0
Skipped Lines 0
Test: Autotune = NA - 
Test: Brownout = GOOD - 
Test: Compass = GOOD - mag_field interference within limits (10.30%)
Max mag field length (637.54) > recommended (550.00)

Test: Dupe Log Data = GOOD - 
Test: Empty = GOOD - 
Test: Event/Failsafe = GOOD - 
Test: GPS = GOOD - 
Test: IMU Mismatch = GOOD - (Mismatch: 0.44, WARN: 0.75, FAIL: 1.50)
Test: Motor Balance = WARN - Motor channel averages = [1445, 1475, 1514, 1547]
Average motor output = 1495
Difference between min and max motor averages = 102
Test: Parameters = GOOD - 
Test: PM = NA - 
Test: Pitch/Roll = NA - 
Test: Thrust = NA - 
Test: VCC = GOOD -

lvale, 2017-05-03:
@weety Please post this support request at the forum, where more people can help you.

http://discuss.ardupilot.org/

closed 2017-05-03 by lvale