ArduPilot/ardupilot #6188: problem in building code

hamididari, 2017-05-04:
Hi guys 
i try to building code but i got this 

```
// BUILDROOT=/home/hamid/ardupilot/Build.ArduCopter HAL_BOARD=HAL_BOARD_PX4 HAL_BOARD_SUBTYPE= TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="9bdf9713" -I/home/hamid/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DNUTTX_GIT_VERSION="1a99ba58" -DPX4_GIT_VERSION="8d505a02" -DUAVCAN=1 -D__STDC_FORMAT_MACROS -DHAVE_STD_NULLPTR_T=0 -DHAVE_ENDIAN_H=0 -DHAVE_BYTESWAP_H=0 -DHAVE_OCLOEXEC=0 -I/home/hamid/ardupilot/Build.ArduCopter/libraries/GCS_MAVLink/include/mavlink -I/home/hamid/ardupilot/modules/uavcan/libuavcan/include -I/home/hamid/ardupilot/modules/uavcan/libuavcan/include/dsdlc_generated
Checking modules
Building /home/hamid/ardupilot/module.mk
%% module_mk
make[1]: Entering directory '/home/hamid/ardupilot/modules/PX4Firmware'
Skipping submodules. NUTTX_SRC is set to /home/hamid/ardupilot/modules/PX4Firmware/NuttX/nuttx/
Generating uORB topic headers
python import error:  No module named em

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

/home/hamid/ardupilot/modules/PX4Firmware/Makefile.make:262: recipe for target 'generateuorbtopicheaders' failed
make[1]: *** [generateuorbtopicheaders] Error 1
make[1]: Leaving directory '/home/hamid/ardupilot/modules/PX4Firmware'
../mk/px4_targets.mk:186: recipe for target 'px4-io-v2' failed
make: *** [px4-io-v2] Error 2

```

####Version
master
#### Platform
[  ] All
[  ] AntennaTracker
[+] Copter
[  ] Plane
[  ] Rover

gmorph, 2017-05-04:
What command are you running to do the build?  From the ardupilot folder with waf run

./modules/waf/waf-light configure --board px4-v2
./modules/waf/waf-light build --target bin/arducopter

Thanks, Grant.

hamididari, 2017-05-04:
Thanks for replying
i type ./modules/waf/waf-light configure --board px4-v2 in teminal and i got this

```
hamid@hamid:~/ardupilot$ ./modules/waf/waf-light configure --board px4-v2
Setting top to                           : /home/hamid/ardupilot 
Setting out to                           : /home/hamid/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : px4-v2 
Checking for program 'arm-none-eabi-ar'  : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-ar 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc 
Checking for HAVE_CMATH_ISFINITE         : no 
Checking for HAVE_CMATH_ISINF            : no 
Checking for HAVE_CMATH_ISNAN            : no 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
Checking for program 'cmake'                   : /usr/bin/cmake 
Checking cmake version                         : 3.5.2 
Checking for program 'ninja, ninja-build'      : not found 
Checking for program 'make'                    : /usr/bin/make 
Checking for program 'python'                  : /home/hamid/miniconda2/bin/python 
Checking for python version >= 2.7.0           : 2.7.13 
Checking for program 'python'                  : /home/hamid/miniconda2/bin/python 
Checking for python version >= 2.7.0           : 2.7.13 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Gtest                                          : PX4 boards currently don't support compiling gtest 
Checking for program 'arm-none-eabi-size'      : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-size 
Benchmarks                                     : disabled 
Unit tests                                     : disabled 
'configure' finished successfully (0.679s)
hamid@hamid:~/ardupilot$ 

```

and type ./modules/waf/waf-light build --target bin/arducopter

```
hamid@hamid:~/ardupilot$ ./modules/waf/waf-light build --target bin/arducopter
Waf: Entering directory `/home/hamid/ardupilot/build/px4-v2'
[14/15] CMake Build px4 msg_gen
[  0%] Built target git_genmsg
[ 33%] Built target git_gencpp
[ 66%] Generating uORB topic multi headers for nuttx
python import error:  No module named em

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

CMakeFiles/msg_gen.dir/build.make:316: recipe for target 'src/platforms/nuttx/px4_messages/px4_actuator_armed.h' failed
CMakeFiles/Makefile2:772: recipe for target 'CMakeFiles/msg_gen.dir/all' failed
CMakeFiles/Makefile2:779: recipe for target 'CMakeFiles/msg_gen.dir/rule' failed
Makefile:481: recipe for target 'msg_gen' failed
make[3]: *** [src/platforms/nuttx/px4_messages/px4_actuator_armed.h] Error 1
make[2]: *** [CMakeFiles/msg_gen.dir/all] Error 2
make[1]: *** [CMakeFiles/msg_gen.dir/rule] Error 2
make: *** [msg_gen] Error 2

Waf: Leaving directory `/home/hamid/ardupilot/build/px4-v2'
Build failed
 -> task in 'px4_msg_gen' failed (exit status 2): 
	{task 140614466892016: cmake_build_task  -> }
['/usr/bin/cmake', '--build', '/home/hamid/ardupilot/build/px4-v2/modules/PX4Firmware', '--target', 'msg_gen']
hamid@hamid:~/ardupilot$ 

```

khancyr, 2017-05-04:
Hello, something is wrong with your python environement. You need to look at miniconda2 how to set it up correctly

OXINARF, 2017-05-09:
@hamididari This issues list is for confirmed bugs or feature requests. For development support please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

I've never used Conda (I didn't even know what it was), but apparently you would need to install Python packages in a different way, not with pip. So either you need to use a standard Python and Pip installation or you need to install Python packages using the Conda method.

closed 2017-05-09 by OXINARF

hamididari, 2017-05-10:
Thanks for replying .
i uninstall conda and install python that fix my problem :)