ArduPilot/ardupilot #6193: Adjust speed during auto mission to match WPNAV_SPEED settings if changed

dellarb, 2017-05-04:
#### Issue details
During an auto mission  there may be a need to change the copter navigation speed from the user. Right now changing the WPNAV_SPEED parameter during an auto mission has no effect on the navigation controller unless a waypoint is changed or the vehicle is cycled out of AUTO into another mode and then back. On ArduPlane changing the equivalent parameter (TRIM_ARSPD_CM) takes immediate effect while remaining in AUTO mode.

We are working on the Medical Express Challenge and looking to see how close we can fly two aircraft (a ArduPlane and ArduCopter) however our inability to change speed for the copter once in Auto is causing many challenges.

#### Version
3.5

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
F450 Flamewheel and SITL Simulator

#### Hardware type
Pixhawk + Simulator

iskess, 2017-05-04:
The channel 6 knob method works well

dellarb, 2017-05-19:
@iskess true however in our application there is no RC (autonomous docking). It seems that there should be an option with telemetry to change copter speed that doesn't involve extra RC hardware?

khancyr, 2017-05-19:
@dellarb You should use the MAV_CMD_DO_CHANGE_SPEED command !

Pedals2Paddles, 2017-05-19:
Yep. In an auto mission you can put DO_CHANGE_SPEED in a mission command at a waypoint. Or you can send MAV_CMD_DO_CHANGE_SPEED at any time I believe.

peterbarker, 2017-12-19:
Given the `DO_CHANGE_SPEED` options, do we need to continually check the `WPNAV_SPEED` parameter for changes?

dellarb, 2017-12-19:
DO_CHANGE_SPEED may work but the behaviour is inconsistent with the Plane software and counter-intuitive to user expectations. Changing a parameter should have immediate effect regardless shouldn't it? I can't think of many other tuning parameters that do not take effect straight away.

peterbarker, 2018-01-09:
Dev call outcome is that yes, changing the wpnav_speed should immediately change the vehicle speed.

rmackay9, 2018-01-09:
I'd like to add though, that for this particular use case, it's better to use the DO_CHANGE_SPEED mavlink command and if that doesn't work for some reason, we should fix it.

osmancns, 2018-01-09:
hello @rmackay9 .  I have same problem. In auto mode mission , when i send  
"MAV_CMD_DO_CHANGE_SPEED " command , pixhawk say that : APM initializing , but SITL copter doesnt react. How can I change drone speed immediatelly during an auto mission ?

peterbarker, 2018-01-10:
On Tue, 9 Jan 2018, osmancns wrote:

> hello @rmackay9 . I have same problem. In auto mode mission , when i send
> "MAV_CMD_DO_CHANGE_SPEED " command , pixhawk say that : APM initializing ,
> but SITL copter doesnt react. How can I change drone speed immediatelly

Can you give me exact instructions on how to produce that "APM 
initializing" message, please?

> during an auto mission ?

DO_CHANGE_SPEED should work right now.

osmancns, 2018-01-10:
hello @peterbarker. I am sorry , it is my mistake. I was sending DO_CHANGE_SPEED with REBOOT_SHUTDOWN commands. So , ardupilot say me APM initializing.

peterbarker, 2018-01-10:
On Wed, 10 Jan 2018, osmancns wrote:

> hello @peterbarker. I am sorry , it is my mistake. I was sending
> DO_CHANGE_SPEED with REBOOT_SHUTDOWN commands. So , ardupilot say me APM
> initializing.

So DO_CHANGE_SPEED doesn't persistently change the vehicle speed.  it's a 
run-time thing.  WPNAV_SPEED is the persistent bit - but you can also use 
mission items to alter that.

lthall, 2018-01-11:
I will take a look at this one.

diludg, 2019-02-10:
> I'd like to add though, that for this particular use case, it's better to use the DO_CHANGE_SPEED mavlink command and if that doesn't work for some reason, we should fix it.

Hi Randy,
During a mission, is it change the maximum speed if the WPNAV_SPEED parameter updated? During the mission how often is it check the WPNAV_SPEED parameter value? When sent the DO_CHANGE_SPEED command, is it automatically set the WPNAV_SPEED parameter value to the speed value sent with the DO_CHANGE_COMMAND?
Thanks.

rmackay9, 2019-02-10:
@diludg,

In 3.6.x (and earlier) changing the WPNAV_SPEED doesn't immediately affect the vehicle's speed.  It only takes effect the next time the vehicle enters Auto mode.  This enhancement request is to make the parameter change take effect immediately.

in 3.6.x DO_CHANGE_SPEED does take effect immediately and it also updates the WPNAV_SPEED which actually isn't good.  In 3.7.0, DO_CHANGE_SPEED will not affect the parameter value.

diludg, 2019-02-11:
> @diludg,
>
> In 3.6.x (and earlier) changing the WPNAV_SPEED doesn't immediately affect the vehicle's speed. It only takes effect the next time the vehicle enters Auto mode. This enhancement request is [...]

@rmackay9 
Thanks for the useful information and so, DO_CHANGE_SPEED command will take effect immediately for the current WP horizontal speed during a WP mission regardless of the WPNAV_SPEED parameter.

closed 2021-09-07 by rmackay9

sulakshanarya, 2022-09-14:
> @diludg,
>
> In 3.6.x (and earlier) changing the WPNAV_SPEED doesn't immediately affect the vehicle's speed. It only takes effect the next time the vehicle enters Auto mode. This enhancement request is [...]

@rmackay9 Can you point me to the commit which implements the DO_CHANGE_SPEED will not affect the parameter value?