ArduPilot/ardupilot #6194: Copter ignoring WPNAV_RADIUS, skipping to next WP too soon

iskess, 2017-05-04:
#### Issue details
When planning a mission using Survey Grid, I check the box "Breakup starts" which adds CAM_TRIGG_DIST commands to stop the camera from triggering during the overshoot, then start it back up again on the next flight line. This has always works well for me using PLANE. However when using Copter, the WP is considered crossed too far before arival, so the camera is shutting down too early. Then the camera starts up again too soon while still in the turn. The other byproduct of this is that it cuts the corner when turning back in. 
I have WPNAV_RADIUS set to 100cm (1m).  Is the WPNAV_RADIUS parameter broken?

[image]

I observed this in actual flying as well as in SITL. 

#### Version
3.6

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk

#### Logs
Reproduced in Mission Planner's built in SITL

rmackay9, 2017-05-08:
Could you try adding a delay into the waypoint (it's the first column of the waypoint command).

iskess, 2017-05-08:
Thanks Randy. Adding a 1s delay doesn't help the camera triggering issue. It still stops the triggering as soon as the next WP is activated, which always happens prior to arrival, depending on the speed at which the copter approaches it. 
Using a Delay does force the copter to not cut the corner by making it stop.  The original WPNAV_RADIUS concept is better because it allows the user to control when the WP is considered captured, and how much the copter will cut the corners. The Delay forces the copter to stop which wastes valuable battery. 
I don't understand what WPNAV_RADIUS is actually doing, if anything.

lvale, 2017-05-08:
@rmackay9 it's the same issue we've been discussing

iskess, 2017-05-09:
I see know that this is related to #6143

rmackay9, 2017-05-09:
@iskess, is it possible that you could test with the latest official release AC3.4.6?  I would just like to confirm if this is a new issue or a long existing issue because of our use of the leash.  I suspect it's the latter.

iskess, 2017-05-09:
This issue was written while using 3.4.6

rmackay9, 2017-05-09:
Ok, so basically this is all caused by the "leash" method we use for traversing waypoints which hasn't fundamentally changed since AC3.0.  Certainly we can do better but it's a significant effort that will mostly likely require input from @lthall (we've been talking about what to do but you know.. everyone has many things on their plates).

By the way, the WPNAV_RADIUS is only used if there's a delay on the waypoint command.  The moment the vehicle is within that distance from the waypoint it will move onto the next waypoint and also do the following do-commands (like trigger the camera).  The WPNAV_RADIUS isn't used at all if there's no delay on the waypoint command.

One workaround may be to use a conditional-delay.  That will delay the start of the next do-command.  Sorry, I don't really have a perfect solution which doesn't require some development.

iskess, 2017-05-09:
Randy, As a work around, what do you think of using CONDITION_DISTANCE = 1m?
Would this wait to shut down the triigering until the copter was within 1m of the WP?
[image]

DonLakeFlyer, 2017-05-10:
Isn't most of the long gap in photos due to the fact that TRIGGER_DISTANCE only triggers after it travels the distance. It does not trigger as soon as it is set.

FYI: The PX4 stack specifically changed their TRIGGER_DISTANCE implementation to trigger immediately instead of only after the first distance is traversed. This leads to better surveys when you are not taking images inthe turns. Since you don't lose the first image when you enter the survey polygon.

iskess, 2017-05-10:
That's a good suggestion and would provide more predictablity. However it is not the cause of the issue here.

OXINARF, 2017-06-05:
@iskess I'm not sure I fully understand if your issue is fixed by latest 3.5 beta release or not, have you tested it?

iskess, 2017-06-08:
No, I haven't tested 3.5 beta yet, but I haven't seen any changes that would have resolved this issue. Randy said the fix will require significant development. This should stay open unless Randy decides otherwise.

OXINARF, 2017-06-08:
You said this was related to #6143, which is merged and in 3.5 since rc5, hence my question.

iskess, 2017-08-12:
@Oxinarf, Sorry I haven't been flying Copter since 3.5 so I haven't been able to test. However it appears this issue is still present in 3.5.1 as being reported in #6757

andreas-eberle, 2017-08-12:
I can confirm, this is still an issue in 3.5.1 (see my issue #6757).

rmackay9, 2017-08-15:
This is the expected behaviour actually so I've added the enhancement label.  A couple of things will help reduce the short-cutting:

- add a 1 second delay onto the waypoint parameter (type "1" into the first column of the WAYPOINT command)
- increase the WPNAV_ACCEL parameter from 100 to 250 to shorten the "leash".
- reduce the WPNAV_SPEED to shorten the leash

lthall, 2018-01-12:
[Related issue that should be addressed simultaneously](https://github.com/ArduPilot/ardupilot/issues/6757)

ddomit, 2020-12-02:
What is the update on this issue @rmackay9 @iskess ? Its very hard to work when the pictures are not following the path correctly

rmackay9, 2020-12-02:
There is a blog post here with an update on the s-curves project: https://discuss.ardupilot.org/t/leash-vs-s-curve-navigation/64745

rmackay9, 2021-06-25:
SCurves is included in Copter-4.1 which has begun beta testing so I think I will close this issue now. txs

closed 2021-06-25 by rmackay9