ArduPilot/ardupilot #6199: Parameter upload extremely slow when done to soon

olliw42, 2017-05-05:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
irrelevant, I believe

#### Hardware type
irrelevant, I believe

#### Logs
not required, I believe

Parameters upload gets very slow when done to soon after a startup. This issue has been seen by two, independently, for AP3.5-rc2 and AP3.6-dev. Seems to be known, and a fix seems to be available, according to hints by Randy. 
For the details see:
* http://discuss.ardupilot.org/t/ap3-6-dev-missionplanner-parameter-upload-slow/17168
* http://discuss.ardupilot.org/t/copter-3-5-0-rc2-has-been-released-for-beta-testing/15713/19

Thx, Olli

rmackay9, 2017-05-05:
Thanks for the reminder.  Yes, the issue is that we simply ignore the parameter request if it arrives before the board is initialised.  What we should probably do instead is cache it and then reply once initialisation has completed.

rmackay9, 2017-05-15:
This is fixed in master and will go out with AC3.5-rc6.

closed 2017-05-15 by rmackay9