ArduPilot/ardupilot #6208: Rover: Need to add support for battery failsafe

gmorph, 2017-05-08:
Rover currently doesn't support battery failsafe.  Not a huge issue for Rovers as they don't fall out of the sky when the battery goes flat but some users have requested it RTL when the battery is low.

mikerob, 2017-05-09:
I think this is more important than you have stated. For IC powered rovers, loss of battery does not necessarily mean they will stop - if the servos stop in place you will hold throttle and steering at whatever setting. 

Is it possible to default it to active and go into "HOLD" to be on the safe side?

Mike

gmorph, 2017-05-10:
We can't default it to active as we don't know the lower voltage threshold.  For a 3S it might be 10.5v but for a 4S it might be 14v.

Thanks, Grant.

WickedShell, 2018-05-25:
This was done in #7213

closed 2018-05-25 by WickedShell