ArduPilot/ardupilot #6210: Tools: change to using waf for building releases

peterbarker, 2017-05-09:
#### Issue details
We currently use the make build system for creating the binray releases for firmware.ardupilot.org.

make is also (soft) deprecated.

We should stop using the make build system for the releases as a prerequisite for removing the make build system

OXINARF, 2017-05-10:
I'll just clarify that only PX4 boards are built using the Make build system, all others are using waf.

magicrub, 2017-05-10:
I think we should add some text suggesting to use waf on any make build with the equivalent command shown. Maybe even a Press Enter To Continue nag prompt to ensure it is seen.

amilcarlucas, 2017-10-22:
Aren't the releases done in waf now ?

magicrub, 2017-10-22:
Peter would be the one who knows... And he opened the issue so I'd say it's still valid.

OXINARF, 2017-10-23:
Yes, this is done, all releases are now done with waf (well, except AVR, but that's another branch).

closed 2017-10-23 by OXINARF