ArduPilot/ardupilot #6212: AP 3.7.1 RFE: Near Crash: RTL takes too long to recover a positive attitude and regain altitude

marcmerlin, 2017-05-09:
@tridge , I wanted to thank you for improving AP so that RTL could be more reliable as a "oh crap" switch to recover from unexpected situations/pilot loss of control. Unfortunately it almost failed me a couple of days ago by losing a lot of altitude before finally recovering.

Video to follow along: https://youtu.be/P5TLpbGoiiE2
I used auto takeoff and that worked ok (45sec)
When I switched to FBWA I had an issue with my controller commanding too much pitch down and while I was trying to figure out what was wrong, I realized I was too low and losing altitude fast, so I switched to MANU, managed to over correct, stall, and then switch to RTL.
0:52:, altitude is 31M airspeed is 10kph (too low).
0:54: Airplane pitches down to -24 degrees likely due to the stall, airspeed is back to 55kph, way above flying speed, altitude is 34M
Now is where I'm not very happy with RTL.
0:55: plane is still pitched down, 10 degrees, still losing altitude and on top of that it decides to turn back down to the RTL point when recovering altitude should be #1 priority (RTL point is 100m, altitude is 32m)
0:57: plane still pitched down 10 degrees, airspeed is now 77kph going towards a crash, altitude is now 14m
0:58 : attitude is finally brought back to 0 towards nose up, altitude is a mere 12m, it's only by luck that the plane didn't crash 
After that RTL does the right thing.

My problem is that RTL lost over 20m of altitude when it could have recovered a positive rate of climb almost instantly from offset 0:54. Sure, only 4 seconds went by, but in those 4 seconds, 22m of altitude were lost and the plane could have crashed into a tree, or the ground if it had started just a bit lower

Plane is a bixler 2, with flaps (not controlled by AP). Flaps were up at the time this happened.
BIN from the flight: http://marc.merlins.org/tmp/bix_near_takeoffcrash.bin

Channels:
o CH01 : right Aileron A
o CH02 : Elevator      E
o CH03 : Throttle      T 
o CH04 : Rudder        R
o CH05 : left aileron
o CH06 : R flap
o CH07 : L flap
o CH12: APM mode

magicrub, 2017-05-10:
it's interesting that your "Alt" is -17m at the begining of th eflight, then drifts to -7m before the hand launch which the actual was around 1-2m. That makes me question all the altitude data but that doesn't excuse the massive altitude loss, just thought I'd mention it.

The root-cause of the problem is the underspeed detection will [hold it in that condition for 3 seconds](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L520) after it has been triggered. I can't remember why it was held for 3 seconds and not, for example, 2 seconds. I can see in your scenario that 3 was clearly too long.

I don't think simply reducing the timeout is the best solution. I think this could use some good discussion but to kick it off, I propose:
- reduce 3 second hold timeout to... 2 second
- reduce the effect proportional to the timeout (ie, at half the timeout we're pitching down half as much as we do now).
- don't pitch down at all if we're recovered, rely on throttle (on motor failure, this causes porpoising.. but maybe that is OK?

marcmerlin, 2017-05-15:
@magicrub it's a pixfalcon, so for all I know the baro sensor in it is not good? That could explain the negative altitude (which I also noticed) although indeed I was by the bay area's bay at an altitude quite near to 0.
Thanks for quickly finding the cause for the long delay before pitching back up. In a regular plane, you do pitch down for more than 3sec even to allow for a good recovery, but then again you don't stall at 30m over the ground either :)
Stupid question (maybe?) : why delay pitch back up once airspeed has recovered, at all?
By the time airspeed has recovered and it's detected by the airspeed sensor and transmitted to the AP, flying speed over the wings should be good again, no? That doesn't mean you pitch up immediately to 15 degrees up from -15 or -30 degrees down due to the G's and possibility to re-stall if you increase the angle of attack too quickly, but unless I'm missing something it should smoothly increase the pitch back to the desired value right away.

In this case, the aircraft had stalled just before I enabled RTL and RTL saw an airspeed of 18kph (too low), so the funny thing is that RTL would have likely recovered more quickly if I had enabled it just one second later when the airspeed was above minimum airspeed again.
The thing is that indeed the airspeed is back to flying speed within one second (or just slightly less), and RTL decreased the pitch down from -34 to 11, so at least it did pitch up somewhat, but then decided that it needed more airspeed for 2sec+ when it could have smoothly gone back from -34 to +10 or whatever pitch without stopping at -11 for a while considering that airspeed was way over the required minimum airspeed or even cruising speed.

Does that match the code you just quoted?
Either way in your options, maybe I'd say "if airspeed is X% over stall speed, do not pitch down to maintain airspeed", this would take care of recovery without a motor: you'd pick up airspeed, pitch back up, airspeed would decay and then you'd pitch back down slowly until the airspeed has reached its target, even if that means nose down because power is gone.

How does that sound?

magicrub, 2017-05-16:
So it turns out the pitch it not being pushed down, it's forcing the throttle/airspeed priority balance to full-throttle. I was mistaken when i said it was pitching down, I think it did that some time in the past but I don't see it in the code now.

The mode it was in is not usually important but it is in this particular. All throttle-controlled modes (FBWA, AUTO, RTL, CRUISE, ...) all use the TECS controller to control target airspeed/altitude via throttle/pitch output. However, you are switching to MANUAL which bypasses all "help" that you're expecting from those modes to the integrators and target altitudes are not being computed. You switch in and out of it - twice - right at a critical time. once at the stall and again just after the RTL pulls you up.

Looking at your video, everything is fine until you switch to MANUAL and pull up hard (causing a stall). You then switch to RTL (while stalled) but your plane is almost freefalling and rolling over onto it's side. Then you switch to RTL but it's pitching steeply down. It *is* pitching you up but it's doing it smoothly while in the stall recovery mode. The integrators were paused while in MANUAL so it did not have the normal reaction time as if you had done that same maneuver in any other mode.

This looks like you panicked and switched to the absolutely worst possible mode. I advise against that :)

However, @tridge and I discussed some ways to allow a smoother recovery but I don't know if they would have helped you in this case.

marcmerlin, 2017-05-16:
@magicrub I'm still not sure we've nailed this one. 
The start of analysis sounds right. The pull up in MANU was sadly due to a bit trim/RC model configuration and the fact that when I turn the knob from auto(#4) to fbwa(#2), I often overshoot and get to manual (#1). RTL has a special oh crap switch to override the mode switch.

Let's go back to 0:52 where things are bad in MANU
0:53 RTL kicks in and keeps control.
It does pitch back up from -34 to -8
0:55 it's at -8 while turning heavily. If it's in stall recovery mode, should it even try to turn?
0:56, it's at -11 while still turning heavily
0:57, it's still at -11 while still turning heavily
We're now a full 3 seconds when the airspeed has been ok.
0:58, pitch finally goes back to  (still losing altitude) while the turn ends

For your explanation, I still don't quite understand why RTL stays at -11 pitch down for so long when it's way past stall and cruise speed. You say "smoothly" but I see it go to -11 and stop there for what feels like too long while making a heavy turn that loses more altitude on top of that.
Also, and that's a separate issue I suppose, if it's still in stall prevention mode, it should absolutely not start to do a turn at the same time, that's the very last thing you want to do. No turn until airspeed is above flying speed and we have positive attitude.

magicrub, 2017-05-16:
I'm curious, why do you switch modes so much? This log has 54 mode changes of which about a third are tiny transient ones. So.. 30ish legitimate multi-second mode changes?

magicrub, 2017-05-16:
Here we are, case closed. Looks like it was flying as best as it could have given the circumstances. I agree when recovering from the stall it should set a very conservative roll limit instead of just letting the roll do whatever. However, I don't think it would have mattered much in this scenario. An action item will be to limit roll during a stall to allow it to level off and relax the wing loading.

TECS_PITCH_MAX=15 limited it's ability to perform a hard pull-up, raising that may have helped, but the airframe was a bit stressed at the time and I'm not sure if pulling up harder would have given a much better result. Hard to tell. Would not have hurt though.
During MANUAL mode the actual pitch spiked up to 60degs, causing the stall but the stall detection does not get triggered until the switch to RTL because the TECS controller is idle. Once you hit RTL you're already in freefall and your nose is dropping down, eventually getting to around -35deg before the pullout takes effect. The red lines are the start/stop of the detected stall condition, it lasts a total of 1.4 seconds and not 3 seconds as we suspected. This assignment [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L542) clears the underspeed flag if the above logic says we're not longer stalling. The 3000ms timer a few lines above that is a max limit, not a sticky heuristic, that will disable underspeed if it's stuck on for that duration. The reasoning is if we can't recover by that time we don't want to drive ourselves into the ground at full throttle.
[image]
.
.
.
It looks like the pitch was simply not responding to the demand for the first little while. It's news to me that the pitch integrator resets on every mode change - Hmm.. that doesn't seem right. I'll open an issue for that. The situation is a bit extreme in this case because you're very very nose heavy: your Pitch PID integrator winds up every time you switch modes to the max value which is 20 (you have PTCH2SRV_IMAX=2000). Setting that higher will allow it to learn the correct value but it's so high that you should really look into the root cause on that one instead of adding a ton of trim. Perhaps the flaps are confusing the system a bit on that one?
[image]
.
.
.
TECS.h = height, TECS.hdem is the height demand (what it's wanting to do). In most cases those should match fairly well. When they diverge that means the plane is going somewhere other than where the autopilot is commanding it to go (generally bad). Green lines are start/stop window of the stall condition. TECS.LF is the load factor. You were clearly pulling a hard turn there and the wing loading went up, requiring a higher airspeed. That's another issue I'll track.
[image]

closed 2017-05-16 by magicrub

EShamaev, 2017-05-16:
Integrator reset on mode change was fixed by @tridge in December I guess:
http://discuss.ardupilot.org/t/reset-of-integral-part-of-pids-with-mode-change/13487

WickedShell, 2017-05-16:
@EShamaev duhr! @magicrub and I were only looking at master, rather then the version of code being flown... That was an obvious mistake in hindsight. Way to much is different on master vs 3.7....

WickedShell, 2017-05-16:
Specifically: https://github.com/ArduPilot/ardupilot/commit/3dcf4cfb55d93e298d2a02732d7eeeb72739f4a7

marcmerlin, 2017-05-16:
@magicrub thanks for the analysis.
Nose heavy: tons of electronics in the nose, plus battery, it's a small bixler2, and it's not mean for all that. It already has some weight in the back, but I just don't want to add too much unneeded weight. I've also noticed that it pitches down pretty violently under power due to how the motor is set in the rear. In other words, I don't need heavy trim to fly, but I need generous elevator up mix when the power is turned on. 
Mode switching, I think it switches quickly enough that I don't see it in the OSD or hear it on the RFD radio link to my phone, so I wasn't even aware of this. I'm not the one switching, but I have a RTL switch that overrides the 6 position mode switch on the transmitter.
I will need to look into it, but it's very possible that the Taranis is sending the first mode from the switch and then immediately sending a correction mode from the override switch.
I don't think it used to do this, maybe there is a new behaviour or bug in the recent openTX versions, that causes this. I will look into it.
Sounds like either way 
a) the behaviour I experienced is improved in master, but not in 3.7.1
b) fixing the taranis not to send spurious mode switches will help until there is a new release that contains the already submitted fix.

All that said, should there be a code change to avoid making turns while in stall recovery mode and/or for a few seconds (3?) while pitch is down and altitude is below target ?
The pilot in me still cringes when I see heavy turns while the plane is trying to break out of a stall and/or recovery airspeed/altitude.
This would allow climbing turns as long as pitch is up and would allow turns on a pitched down airplane with a dead motor, after 3 seconds.

marcmerlin, 2017-05-16:
@magicrub as a thanks for your work, I'll share another short video where AP did beautifully in very unexpected conditions.
I got a fairly big branch that got attached to my plane's T tail and that I didn't see until the plane flew away with the branch attached (totally different aircraft)
https://youtu.be/bjkU3Q8VlwQ?t=13s
Auto takeoff did great despite the big drag and big CG upset.
Things went a bit pear shaped once I went to FBWA and cut the power, which caused the branch to drag the plane down, stall it and almost invert it.
My low altitude GUIDED protection kicked in quicker than I was able to react and come up with a plan, guided took back over, from a 110 degree roll, fixed it, went back up and picked up some altitude. I finally took back control in FBWA but didn't leave enough power in to have full control and stalled over the ground despite an airspeed of 62kph (should have kept more power to offset the drag)
Either way, AP did really well considering a very unusual flying configuration!

marcmerlin, 2017-05-16:
BTW, @magicrub I've been wanting to run mavexplorer against the logs myself to verify the mode switch problem, but sadly my fully working mavexplorer install broke after I upgraded something else.
Trying to fix it has been an exercise in wasted time and frustration :(
For now I filed https://github.com/ArduPilot/MAVProxy/issues/413
If you happen to know of a python archive of mavexplorer, let me know.

magicrub, 2017-05-16:
I've flown my big Skywalker x8 with a pair of 25 foot streamers attached. Still flew perfectly and even had an amazing pin-point auto-land :) In the older code if you don't switch modes the PIDs make up for all kinds of abnormal drag.
https://www.youtube.com/watch?v=bPbg3s6EOJc

marcmerlin, 2017-05-17:
@magicrub now that I have my MAVExplorer working again, I figured out why you saw so many mode switches in the log.
Sadly, due to the ESC init bug, I left my plane "running" between 2 flights and the logs show a previous flight, before showing the actual flight described in the video I linked.
Logs do start at
Tue May  2 16:27:11 2017 MAV.flightmode=MANUAL       (MAV.timestamp=1493767631 0%)
however that's not the flight in the video. My apologies for not having a working mavlink log analysis setup when I filed this issue (fixed now)

Once I trimmed out the irrelevant flight, there are only 12 mode changes in the flight:
Tue May  2 17:03:33 2017 MAV.flightmode=AUTO         (MAV.timestamp=1493769813 88%)
Tue May  2 17:04:07 2017 MAV.flightmode=CRUISE       (MAV.timestamp=1493769847 90%)
Tue May  2 17:04:09 2017 MAV.flightmode=FBWA         (MAV.timestamp=1493769849 90%)
Tue May  2 17:04:10 2017 MAV.flightmode=CRUISE       (MAV.timestamp=1493769850 90%)
Tue May  2 17:04:12 2017 MAV.flightmode=MANUAL       (MAV.timestamp=1493769852 90%)
Tue May  2 17:04:15 2017 MAV.flightmode=RTL          (MAV.timestamp=1493769855 90%)
^^^ this is where I start RTL for recovery

Tue May  2 17:04:21 2017 MAV.flightmode=MANUAL       (MAV.timestamp=1493769861 90%)
^^^ this is where RTL managed to recover and not crash despite losing a bunch of altitude, and I took back over control.

The only reason I mention this is that you said I did multiple flips from RTL to MANUAL and that it reset integrators. The logs don't seem to show this.
(I also spent time checking my taranis config to make sure it was not sending spurious sub second mode changes, and turns out I had already fixed that a long time ago).
So I'm still willing to believe you if you say 3.8 already fixes the problem I saw, but your explanation about integrators being reset because I switched out of RTL mode for sub second times, doesn't seem to match the logs (or I misunderstood you)

Below is the rest of the flight, uneventful vvv
Tue May  2 17:04:36 2017 MAV.flightmode=FBWA         (MAV.timestamp=1493769876 91%)
Tue May  2 17:06:08 2017 MAV.flightmode=RTL          (MAV.timestamp=1493769968 95%)
Tue May  2 17:06:16 2017 MAV.flightmode=FBWA         (MAV.timestamp=1493769976 95%)
Tue May  2 17:06:36 2017 MAV.flightmode=MANUAL       (MAV.timestamp=1493769996 96%)

magicrub, 2017-05-17:
This is your Pitch PID Integrator value. In your first flight you'll see that it grows to the max and goes in and out of max (2000). On each mode change that integrator memory is erased (6 months a few weeks after 3.7 was released). The pitch integrator is important because if it gives a PITCH UP demand, it expects to achieve it. If it doesn't then it raises the pitch integrator so that it pushes the pitch harder. And repeat. When it switched to RTL it was very clearly pulling up on the pitch, it just wasn't pushing hard enough... and it was learning to push harder over the next couple seconds. Then you switch to Manual and it zeros again and stops tracking it. Then it goes to FBWA and starts learning it again over the next few seconds until it clips at 20. If your integrator was already maxed when it switched to RTL then it would have pulled up much harder and not pointed to the ground much at all.
https://cloud.githubusercontent.com/assets/4782875/26091926/03d25e22-39c3-11e7-9fda-03243b3138bc.png

marcmerlin, 2017-05-22:
As per your previous comment, I didn't realize that simply switching once from FBWA to RTL reset the integrator that takes several seconds to reset/relearn. I thought you meant I had switched in and out multiple times when in this case it was left in RTL 8 seconds, almost 5 seconds during which it wasn't pitching up when it should have been.
Glad to hear that it's been addressed in 3.8

As a reference/related issue, this is the discussion we had over a year ago about not pitching down to gain airspeed unless airspeed is below FBW_ARSPD_MIN + X% 
https://github.com/ArduPilot/ardupilot/issues/3582
which is still an open issue, but I'm not sure if it's been addressed in AP 3.8 or not.