ArduPilot/ardupilot #6213: RFE: Allow curve for current mapping from power module

marcmerlin, 2017-05-09:
As explained here
http://discuss.ardupilot.org/t/could-ap-allow-for-non-linear-current-sensors/17244
many current sensors aren't really linear, even though they should be.

I took 3 power modules, a chinese one rated for 150A, and 2 3DR ones, rated for 90A. The 3DR ones were identical looking but not from the same order, and didn't have the same amp per volt mapping or even the same volt multiplier either. Not a huge deal, there are;
- BATT_AMP_PERVOLT
- BATT_AMP_OFFSET

However, none of my 3 current modules show any current when actual current flowing out when the plane is idle (0.5A), so I set BATT_AMP_OFFSET to -0.02 to get 0.5A when ardupilot is reporting 0 current
However, this does not address the fact that the ratio of Power Module Voltage Present on the I pin (PM V in the doc below) divide by the actual Amps seen out of the battery (with a real amp meter wired between the battery and the power module), labelled as A in the table below.
The end result is the A/V value which starts low, and rises as amps go up, and that's true for all 3 sensors. The end result is that it's just going to be wrong to have a single A/V value and it should be at least a 5 point curve to give a reasonably accurate value:
https://docs.google.com/spreadsheets/d/1sLU_pinxh7hyYVzwJxjreoqXru3tXWDeAOFb7Krs_2Q/edit?usp=sharing

The table above also has a "AP A@ xxx A/V" value, which amp value displayed by apm planner after I put in the best BATT_AMP_PERVOLT I could come up with.
What's weird, is for the 2nd PM, the apmplanner displayed values actually ended up being close to the real amps coming out of the battery, despite it not being linear. I'm not sure why.
The 3rd PM however, I didn't get any good amp values for amps below 6A or so, and no amps at all for anything below 1.7A, which is weird considering it's on the same flight controller, and with 2 different genuine 90A 3DR power modules that look identical.
But the point is that if ardupilot supported a I/V curve for power modules, I could get a much more accurate current value regardless of the sensor.

lvale, 2017-05-10:
@marcmerlin Can you try real linear power modules like the ones from Mauch? The kind of power modules you tested can vary a lot depending even on environmental conditions, so if a polynomial curve would be done for a specific module, that same curve for the same specific module would be wrong even if the ambient temperature is different.

Pedals2Paddles, 2017-05-10:
I agree this may be an exercise in futility. The shunt type power modules will give readings all over the map depending on numerous variables. These are hardware limitations that you cannot fix for in software. I believe the solution is live with that level of performance, our use a better power module that doesn't suffer from those issues.

iskess, 2017-05-12:
I agree that the Shunt type is inherently less accurate, and I use the Mauch sensors myself.  However, the APM project has always had much less accurate current sensing than others. I've used the old Dragon DOSD many years ago, the RangeVideo RVOSD, Cyclops OSD, and others I can remember the name of. These were all much more linear and accurate and all used the shunt type of current sensor. It has improved since we moved from the APM 2.x to the Pixhawk, but it is still notoriously inaccurate. 
I don't know why this is, perhaps a software improvement is worthwhile. While using a shunt sensor will never be super accurate, allowing a curve to be programmed would make a big difference. Even if there were only 3 points on the curve so we can plot low power, cruise, and max throttle.  
99+% of the current sensors being used are the shunt type, so it would be good to  accommodate them.  Far from perfect, but not futile.

EShamaev, 2017-05-12:
Shunt sensors are much more linear than any hall sensor will ever be. Other issue is how you use it. A simple instrumental opamp will make huge difference.
Even this little thing: https://www.sparkfun.com/products/9816 can improve performance of you sensor a lot, however it requires some tweaks on components (resistors change) if used with our common current sensors.
Question is, should we fix hardware issues of extremely cheap current sensors?

marcmerlin, 2017-05-15:
@lvale I can change all my power modules if I must, but I paid good money to get them from 3DR with my pixhawk and would rather be able to use them :)
Now, if they change a lot with temperature, so be it, but at least, it would be good to have numbers match on my bench to some extent.
One reason why I do care is actually safety in the end: having some idea of how many mAh got used is kind of important to know how much runtime I have left (at least for a plane where you don't fly by time if you have a motor glider or generally don't need to fly at full or even partial power all the time)
@iskess generally I agree with what you said, I'm not looking for ultra perfection, but for something better than "extremely bad" which is what I seem to be getting right now :)
That said, sometimes hardware sucks beyond repair too, I have a pixfalcon with a real 3DR power module and it will give me different voltages for the battery depending on whether I reboot it or not. That's not something that software can fix, but in better cases (real pixhawk with real 3DR power module, those sold plenty), having a curve to give better numbers for those, would be good.

lvale, 2017-05-15:
@marcmerlin I also have lousy power modules and use them. The compromise I did to get reasonable results on these "less than optimal" power modules, was to calibrate them to the "normal" operating current, and did some measurements to total capacity reported by the logs vs total charged capacity by the battery charger and adjusted the parameters.

EShamaev, 2017-05-15:
The best shunt current sensor so far is made by @proficnc and is supplied with pixhawk2: http://www.proficnc.com/power/80-power-module.html . It has *high* side shunt with a special IC for high side current measurement. As long as you will have shunts on low side you will suffer from all possible issues and non-linearity.
There also exist power module from 3DR with almost the same design (high side shunt) with INA169 (but I have never used it) and if you are using this one, it is very strange that you have issues. They should provide +/- 5% over whole temp range with very good linearity.

marcmerlin, 2017-05-15:
@lvale I know how you can aim for the middle so that the current reading is not too far off at full throttle and not too far off at idle, but that's not good enough IMO.
I have a plane where idle shows 0 amps and I had to fudge with the amp offset at 0 setting to artificially set the 0.6A or so my plane uses when idle.
You might not care, most probably do not, but I do. The amp value will never be perfect but I look at it like a fuel flow gauge in a real plane: My only way to know how much fuel I have left in the tanks is to know my tanks were full when I left and how much fuel went through the fuel flow sensor. Anything else is bogus.
Same thing for lipos (given that I cycle the lipo from time to time to re-measure its discharge capacity). I lost a vehicle in an expensive crash by having a lipo die at 66% discharge while I was landing. It really is important to know how much current was used during a flight, as closely as possible.
The whole discussion here has been how there are better sensors. Sure, there are, but this request was to allow for a better way to get more usable values from existing sensors that most everyone has with a multi point curve. Honestly a 3 or 5 point curve would be enough to give significantly better results IMO, which is the point of this RFE issue.

marcmerlin, 2017-05-15:
While this is obviously not a new issue or a bug per se, it is ultimately a safety issue not to better know how much power was actually consumed from the battery, so I've added the safety label.

Pedals2Paddles, 2017-05-15:
As far has #safety is concerned, if it is such a critical safety issue, using inferior hardware that cannot accurately report the current flow seems to be the problem. Replacing the inferior and apparently dangerous hardware would be the correct solution.

You're asking for a software solution to poorly designed or inappropriate hardware.  That is never appropriate. And given the problem varies with environmental conditions and flight conditions, it is not only inappropriate, but completely impossible to do right.

I get where you're coming from. I've used fast and cheap power modules too. But as the saying goes, you get what you pay for. In pay for better now that I know better.

marcmerlin, 2017-05-15:
@Pedals2Paddles I would usually totally agree with you, but keep in mind that I'm not complaining about some crappy clone I bought from HK.
I'm talking about the default 3DR current sensor that was sold in some unknown of thousands of units (tens, hundreds of thousands?) and that power heavy multirotors, some are lethal if they fall on people (I know I bought one from them, that's the one that got the dead battery in the air and almost a made a hole in the empty parking lot pavement in the place it crashed. There was no one around (by design), but it could have been much worse).
Now I cannot say that it is not a failure of the lipo over the current sensor having told me I used 4200mAh out of 6000mAh. The lipo was fully charged and verified before takeoff, the current sensor was calibrated a few days prior. I just don't know whose fault it is and can't say after the fact, but knowing that the current sensors can return A used values that can be totally wrong by a factor of 1.5 or even 2 (in either direction), doesn't feel me with fuzzy feelings.

Now, I mostly only fly foam planes and I don't overly care if I lose power because I can still control the plane with a dead motor, but others are still flying RC gear that's much more lethal than mine.
Saying that everyone should replace their hardware (the one that has been sold for years) to be safer when there is a software workaround that can improve the data reliability a fair amount, I'm not convinced is the right answer :)
(especially for all those 3DR multirotors where all the guts are inside and hard to get to/replace)

marcmerlin, 2017-05-15:
Your other point that it varies with temperature and all, is valid, but it's not because it can't be done 100% accurately that there is no benefit in going from 50% accurate to 85-95 accurate.

magicrub, 2017-05-15:
The nice thing about supporting "crappy hardware" is it forces the driver/implementation to be more robust.

Pedals2Paddles, 2017-05-15:
That is the same one I used to use. Lived and learned. Just because 3DR sold it doesn't mean it is a great product.

So you're basing this on a crash you're assuming was caused by inaccurate mAh used data caused by the power modules hardware limitations. But could just as easily have been caused by the LiPo not performing perfectly. Have you tried to repeat the scenario on the ground to see how it actually plays out? It should be pretty easy to replicate on the ground.

Were you not monitoring the voltage? That is way more useful than mAh remaining when using an inferior power module in my experience. And it works regardless of any other inconsistencies. Voltage should have been in alarm long before your copter became a smoking hole in the ground. Was that ignored? Not set? 

I'm not saying the solution you're proposing doesn't have a useful purpose. If it existed today, I would use it on mine that have those crappy modules still. But you will be hard pressed to find a busy ArduPilot developer to reengineer the whole thing for what at the end of the day, is a software solution to inferior hardware. If you or someone actually did it, I would totally support merging the feature in. But good luck finding someone to actually do it.

marcmerlin, 2017-05-15:
@Pedals2Paddles you misread what I wrote. I said it could have been the current sensor, it could have been something else like a bad lipo. I have no way to tell.
I have however replicated very non linear current sensor readings on the bench with proper test equipment, see my first post and the spreadsheet I posted.
Yes, I was monitoring voltage, but:
1) when you're drawing 40-50A, voltage sags quite a bit, so it's hard to use that. I've had 4S lipos routinely show 12.5V (out of 16.8V) for various reasons on takeoff. Note too that not only the lipo will sag, but the voltage will sag a bit despite the thick wires, by the time it reaches your sensor
2) by the time the voltage is actually falling visibly, it's too late if you need 1-2mn to get back to the landing spot and do a slow descend/proper landing (what happened here is the voltage sagged visibly after the aircraft was already in landing phase, so there wasn't much more that could be done)

Your last point about "good luck finding someone to do it", sure, I'm just filing an RFE and that doesn't magically allocate someone to do the work, especially unpaid. It may never happen as a matter of fact, that's totally how things work, but without an RFE to start with, then it's even more guaranteed to never happen :)
At least I put in a few hours of work testing multiple current sensors and posting a spreadsheet with results in case it helps.

lvale, 2017-05-16:
@marcmerlin While a accurate current measurement gives us a estimate of available flight time, it's not a accurate measure of flight time available.
As you know the batteries can have different behaviours depending on a series of environment and usage pattern, and the feared battery dump where the voltage drops abruptly can not be estimated / guessed by the most accurate current sensor.
With our current systems (hobby grade) there are ways to improve the estimate of flight time/security, and not using the consumed current.
Per cell voltage measuring is perhaps the easiest way, and although ArduPilot doesn't have mechanisms in place to report/log per cell voltage measuring, with some 3rd party systems (see FrSky http://alofthobbies.com/frsky-sp-flvs-smart-port-lipo-voltage-sensor.html) you can have a accurate state of the cells.
While I see the benefit of having a more accurate current consumption (see https://www.amploc.com or http://www.mauch-electronic.com/50a-100a-200a-hall-sensor) this would provide users with a false sense of safety/security. I've had my share of batteries dumping because I either use them differently or weren't correctly charged or simply were old and the correct current consumption was well within the "usable" range but one/more of the cells went below 3.2V, not enough to trigger a low voltage warning but enough to damage the cell.

We've had discussions to include a better estimate of battery state before, and a good current reading would be helpful to add to that estimate, but the complexity of calibrating any current sensor with a n degree polynomial would be beyond the reach of the vast majority of the users, that would resort to "default" which is what we have today.

marcmerlin, 2017-05-16:
@lvale thanks for the detailed answer.
For that it's worth I actually own the frsky lipo per cell sensor, I just can't use it in a useful fashion since pixhawk/AP does not support frsky sport sensors (I also have an frsky RPM sensor I'd love to use)
You are correct that a cell can suddenly drop below 3V and cause clear issues, although in my specific case, the 4S lipo was showing 5V (last reading sent before the pixhawk rebooted on undervoltage) just before it crashed into the ground.
However, I get your general point that doing a 3 or 5 point calibration would be extra code, non trivial for many users, and therefore they'd be unlikely to do it in the end, making the extra code mostly useless.
If you think nothing will come out of this issue, feel free to close it.

EShamaev, 2017-05-16:
Just out of curiosity, can you share your setup for measuring current on battery and resulting output from power module?

marcmerlin, 2017-05-16:
On Tue, May 16, 2017 at 07:12:31AM -0700, Eugene Shamaev wrote:
> Just out of curiosity, can you share your setup for measuring current on battery and resulting output from power module?

Sure, I have a 20A multimeter with a custom set of wires that connect it
inline with some XT60 plugs. 
I put that between the battery and current sensor.

I then simply turn on the motor with the aircraft on the bench to start
drawing more amps while reading the amps/volts from the battery and reading
them in apmplanner and adjusting the coefficient until they match.

Now, I have a chinese 600W battery discharger sitting in a corner somewhere,
which I could actually connect instead of the ESC to smoothly draw more amps
without spinning any motor. Maybe I should try that next time.

Marc
-- 
"A mouse is a device used to point at the xterm you want to type in" - A.S.R.
Microsoft is to operating systems ....
                                      .... what McDonalds is to gourmet cooking
Home page: http://marc.merlins.org/

EShamaev, 2017-05-16:
I would like to investigate it, is it possible to ask you to make same experiment with 3DR's 90A power modules, but take readings not from Ardupilot, but just voltage with DMM without autopilot?

marcmerlin, 2017-05-16:
@EShamaev actually I described my setup wrong. I did also read the I voltage from the PM with another volt meter and logged that. See the spreadsheet linked at the top
https://docs.google.com/spreadsheets/d/1sLU_pinxh7hyYVzwJxjreoqXru3tXWDeAOFb7Krs_2Q/edit?usp=sharing
PM V is voltage on the power module on the I trace
A is amps on my multimeter coming out of the battery
A/V is the obvious ratio between the 2 showing the non linearity.
AP A@xx is the amp reading shown by ardupilot with the ampspervolt value noted

rsilk1949, 2017-07-03:
Since most power modules are notoriously inaccurate when it comes to measuring voltage, I installed a FLVSS voltage sensor which reads the voltage of all cells directly from the battery via the balance plug.
There are excellent instructions on setting up Mission Planner with the Taranis/OpenTX and Turnigy 9XR Pro on the Adupilot docs. You will need X series receiver/module and C&T telemetry cable.
Now I can see total, lowest, individual cell levels on the radio and set repeating voice alerts.

marcmerlin, 2017-07-03:
@rsilk1949 ideally measuring voltage should be pretty easy and accurate compared to measuring amps.
However, I already have the per cell lipo frsky sensor, it cannot feed data into ardupilot. That's tracked by 
https://github.com/ArduPilot/ardupilot/issues/6296

julled, 2023-01-17:
I support this request. I think its not a lot of software complexity added and a lot of safety could be gained for cheap setups.

lthall, 2023-01-17:
Sounds like the best solution to this problem is to make a LUA Battery type. That way people can do what ever they need to do without restriction.

julled, 2023-01-18:
> Sounds like the best solution to this problem is to make a LUA Battery type. That way people can do what ever they need to do without restriction.

thats a nice suggestion! i am wondering if the speed of the lua script can be fast enough that i can integrate the current consumption?