ArduPilot/ardupilot #6221: Copter: FrSky Passthrough Telemetry - ap_status offsets wrong/missing?

Clooney82, 2017-05-10:
I´m currently working on a new lua script for FrSky SPort Passthrough (OpenTX) protocol (X-receivers)
Therefore I found 2 things I supposed to be wrong:

**First:** As described in AP_Frsky_Telem.cpp: uint32_t AP_Frsky_Telem::calc_ap_status(void)
5 bits should be used for flight modes and 2 bits for Simple & SuperSimple 
but SSimple bits are only shifted 4 bits left, not 5 (**line 676**). Therefore it will overwite 1 bit of the flight modes reserved bits.

**Second:** For me it also seem like shifting of AP_LANDCOMPLETE_FLAG (should be shifted 7 bits left) is missing.

BR, Jochen

floaledm, 2017-05-10:
Thanks for taking the time to provide feedback. I double checked and those you mentioned are fine. You should read more about ap_valuep as found in the init function. The shifts are relative to the bits found in ap_valuep...

magicrub, 2017-05-10:
@floaledm this is your domain. Is this ready to be closed?

floaledm, 2017-05-10:
Yes, sorry about that. Thanks for the followup Tom!

closed 2017-05-10 by floaledm

Clooney82, 2017-05-10:
@floaledm thanks for the feedback. You´re right, I missed the part with ap_valuep.

floaledm, 2017-05-10:
Let me know if you find anything else... don't hesitate to contact me directly with questions.

michelsneto, 2017-08-29:
@Clooney82 , Hi I´m very very newbie on these subject and no deep experience on this programming structure for copter project or lua scripts.
I´m studing some lua scripts based on Teensy or other Arduinos telemetry converters trying to translate these lua scripts to use with this new FrSky SPort Passthrough (option 10 on copter) with no additional hardware (other than simple logic inverter an diode).

I did set and put to work and watch sport pin on X8R with a logic analyser and I could see the x8r receiver making pooling for sensors (0x7E...) and pixhawk replying on Physical ID 28 (0x1B) with logig IDs 0x5000; 0x5001; 0x5006 and 0x5007; but I can´t detect these new sensors on Taranis scanning.

Is there any tip for this or any special lua script ? If yes, could you, kindly,  indicate whre can I download it to try ?