ArduPilot/ardupilot #6222: Another Windows 10 Build Issue

maroquio, 2017-05-10:
#### Issue details
After install Cygwin, when I run:
$ make sitl -i
../mk/apm.mk:25: /home/Ricardo/ardupilot/modules/uavcan/libuavcan/include.mk: No such file or directory
make: *** No rule to make target '/home/Ricardo/ardupilot/modules/uavcan/libuavcan/include.mk'.  Stop.

#### Version
Using last version.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quadrotor

#### Hardware type
Navio2

#### Logs
No logs yet.

Any help?

EShamaev, 2017-05-10:
Did you do ``git submodule update --init --recursive`` ?

magicrub, 2017-05-10:
please don't "make sitl", we really need to retire make completely. We suggest using waf for builds and for SITL use the script: sim_vehicle.py

maroquio, 2017-05-10:
Yes, I updated git submodules...

magicrub, sorry, but It's my first time with SITL under Windows. What do you mean with "waf for builds"?

magicrub, 2017-05-10:
you don't build SITL, you run it. Just navigate to ardupilot/ArduPlane or ardupilot/ArduCopter folder and type "sim_vehicle.py". If that doesn't work, you need to follow the wiki instructions
http://ardupilot.org/dev/docs/sitl-native-on-windows.html

OXINARF, 2017-05-10:
@maroquio This issues list is for feature requests and confirmed bugs. For development question please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

The error looks like you haven't updated the submodules properly using the command @EShamaev gave you. Check that you have the file `modules/uavcan/libuavcan/include.mk`, if you don't, then you haven't updated the submodules correctly. If you still have problems do what I said above and we'll try to help you out.

closed 2017-05-10 by OXINARF

geraldlow7, 2018-03-01:
Hey, i have the same error and I followed the instructions to the dot.

I also checked that the file modules/uavcan/libuavcan/include.mk exists.

This is the error i get when i try to make sitl -j4:
../mk/apm.mk:25: /modules/uavcan/libuavcan/include.mk: No such file or directory
make: *** No rule to make target '/modules/uavcan/libuavcan/include.mk'.  Stop.

This is the error I get when I run sim_vehicle.py -j4:
...
RiTW: Window access not found, logging to /tmp/ArduCopter.log
SIM_VEHICLE: MAVProxy exited
SIM_VEHICLE: Killing tasks