ArduPilot/ardupilot #6225: SITL Parameter system does not work.

SwiftGust, 2017-05-11:
#### Issue details
default parameter setup in the SITL does not work.
When I switch frames using
sim_vehilce.py -v ArduCopter -f gazebo-iris
sim_vehilce.py -v ArduCopter -f gazebo-X2 (custom)
does not changes any parameter between two launches.
(I added mine in the sim_vehicle.py)

#### Version
Copter-3.5 rc 5 in the current master

#### Platform
[  ] All
[  ] AntennaTracker
[V] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad-X(Iris / My own Coax Copter)

OXINARF, 2017-05-11:
As talked in Gitter this works. You should realize though that the parameter file is a defaults file so if you **write** a parameter and don't wipe the settings after, then the saved setting will be used as it has priority over a default.

closed 2017-05-11 by OXINARF

SwiftGust, 2017-05-11:
Well, Actually It is not working properly. with wiping(-w) some of things get clean and reloaded. but not everything. 
For example I have set rc channel min max 1000/2000, but it shows 1100/1900 in SITL. and so on...

gmorph, 2017-05-11:
Hi SwiftGust.  When you start SITL it loads a default set of parameters from this directory:
ardupilot/Tools/autotest/default_params
I bet that's overriding your own local settings.

Thanks, Grant.

SwiftGust, 2017-05-11:
Hi, Thank you for reply.
I made own parameter in ardupilot/Tools/autotest/default_params
and also launch command in sim_vehicle.py.
Look like something messed up with my desktop. all others tested it and says it working.