ArduPilot/ardupilot #6227: How to change udp Bind Address?

AliSiddiqui4, 2017-05-11:
Hi, 

So I am want to simulate multiple Erle Copters in Gazebo. I am able to create multiple instances of MAVlink by changing sim_vehicle.sh. Then rosrun the launch in which I have 2 groups where I am defining the erlecopter. Everything seems to be fine the only problem is that I am unable to change the udp bind address can someone help? 
```

`process[gazebo_gui-7]: started with pid [28833]
[ INFO] [1494527178.857719148]: FCU URL: tcp://127.0.0.1:5760
[ INFO] [1494527178.859154560]: tcp0: Server address: 127.0.0.1:5760
[ INFO] [1494527178.859893693]: GCS URL: udp://127.0.0.1@127.0.0.1:14550
[ INFO] [1494527178.860557831]: udp1: Bind address: 127.0.0.1:14555
[ INFO] [1494527178.861373563]: udp1: Remote address: 127.0.0.1:14550
[ INFO] [1494527178.907755780]: FCU URL: tcp://127.0.0.1:5770
[ INFO] [1494527178.909053775]: tcp0: Server address: 127.0.0.1:5770
[ INFO] [1494527178.909740242]: GCS URL: udp://127.0.0.1@127.0.0.1:14560
[ INFO] [1494527178.910412339]: udp1: Bind address: 127.0.0.1:14555
[ INFO] [1494527178.910704767]: udp1: Remote address: 127.0.0.1:14560
[FATAL] [1494527178.925428837]: GCS: DeviceError:udp:bind: Address already in use
spawn_model script started
spawn_model script started
[ INFO] [1494527179.145939933]: Plugin 3dr_radio loaded and initialized
[ INFO] [1494527179.151043515]: Plugin actuator_control blacklisted
[ INFO] [1494527179.155162080]: Plugin altitude blacklisted
`
```

khancyr, 2017-05-12:
Hello, we are not supporting Erle Robotic plugin, so it is better to ask them directly on their forum.

OXINARF, 2017-05-16:
@AliSiddiqui4 This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

 This doesn't look to be an ArduPilot issue. That log looks to be from Gazebo and as you can see the remote addresses all have different ports, it's the Gazebo side that is trying to bind to the same port.

As @khancyr said, we don't use the Erle plugin, look for our instructions in our wiki: http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html

closed 2017-05-16 by OXINARF