ArduPilot/ardupilot #6233: Segmentation fault in Linux RCInput_RPI

patrickpoirier51, 2017-05-12:
#### Issue details
RPI , Linux based FC that use GPIO for PPM input is crashing with this error: Segmentation fault in Linux::RCInput_RPI::_timer_tick() 

#### Version
3.4.2  -  3.5

#### Platform
[  ] All
[  ] AntennaTracker
[x  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_N/A)_

#### Hardware type
_RPI GPU based inteerupt handler like PXF'MINI_

#### Logs
_Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 774]
0x000b8974 in Linux::RCInput_RPI::_timer_tick() ()
(gdb) bt 0x00119730
0x000b8974 in Linux::RCInput_RPI::_timer_tick() ()_

Might be indirectly reated to: https://github.com/ArduPilot/ardupilot/issues/5226
Works OK with: V3.4-dev (5362dff5)

rmackay9, 2017-05-13:
Thanks,  I'm pretty sure we've had another report as well that the ErleBrain isn't working on AC3.5

patrickpoirier51, 2017-05-13:
You might be referring to this: 
http://discuss.ardupilot.org/t/arductoper-3-5-not-working-on-pi-zero/17105

lucasdemarchi, 2017-05-19:
I think I nailed it down to the GPIO_RPI class... probably doing simultaneous access to the HW and causing problems? On a test here I just replaced that class with Empty::GPIO gpioDriver. It's running for some time now, way more than before.

Not sure what change on GPIO_RPI caused this, though.... I'll leave it running for some more time.

patrickpoirier51, 2017-05-19:
As a side note, with the new code I could not get the MS5611 started. I changed with a new sensor and it started. Try to play with parameter on the SPI Bus speed  https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/SPIDevice.cpp#L94  
but could not get the old MS5611 started,  but it still work fine with old release.

lucasdemarchi, 2017-05-19:
MS5611 appears fine for me with a RPI3 + navio+. I left it running for ~7h and it's working.

closed 2017-05-22 by tridge

onceme, 2017-08-28:
> RPI , Linux based FC that use GPIO for PPM input is crashing with this error: Segmentation fault in Linux::RCInput_RPI::_timer_tick()

I think inside _timer_tick(), the empty DMA control block address would cause Segmentation fault as well.  Could you reproduce it after applied #6301? If yes, please try #6755, thanks.

patrickpoirier51, 2017-08-28:
I did not experienced occurences since, but I am not using this FC very much.
@LanderU , from Erle shoud test this because of its user base he might have enough sampling to confirm your patch. (that seem to address the core of the problem)

LanderU, 2017-08-28:
Is working for my after applying this: #6301

PR #6301 (2017-05-22): Fix RPI boards
@patrickpoirier51 @LanderU @staroselskii can you give this a try?

Should fix #6233