ArduPilot/ardupilot #6234: Rover: MAV_MODE_FLAG_SAFETY_ARMED not set accordingly

jaxxzer, 2017-05-12:
#### Issue details
All arming checks disabled, BRD_SAFETYENABLE enabled:
arm vehicle, base_mode flag does not change to reflect armed status.

All arming checks disabled, BRD_SAFETYENABLE disable:
Disable BRD_SAFETYENABLE, arm vehicle, base_mode flag changes to reflect armed status.

#### Version
master: 46cba2c9

#### Platform
Rover

#### Hardware type
Pixhawk1

gmorph, 2017-05-13:
How do you know its not being set?  Have you checked a log and done a bit operation to confirm or something?

In GCS_MavLink.cpp we have this code in send_heartbeat (note the comment isn't quite right but the code is)

    // we are armed if we are not initialising
    if (control_mode != INITIALISING && hal.util->get_soft_armed()) {
        base_mode |= MAV_MODE_FLAG_SAFETY_ARMED;
    }

So providing we aren't in the INITIALISING mode we should always be setting the base_mode flag correctly.

Thanks, Grant.

jaxxzer, 2017-05-13:
I'm looking at the values in the HEARTBEAT fields. I don't have a rover, I was looking at this while looking into how QGC determines if a vehicle is armed. The main LED turns solid blue, a statustext message says 'throttle armed', but the base_mode flag isn't sent. I think the issue is when you hal.util->set_soft_armed.

Here's two .tlog files taken with QGC.
[brd_safety_enabled.txt](https://github.com/ArduPilot/ardupilot/files/998287/brd_safety_enabled.txt)
[brd_safety_disabled.txt](https://github.com/ArduPilot/ardupilot/files/998288/brd_safety_disabled.txt)

OXINARF, 2017-05-17:
This is because of this line: https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/APMrover2.cpp#L153

It only sets armed state if the safety switch was pressed when enabled. Plane does the same, but Copter and Sub don't. I'll mark this as dev call topic so we can discuss.

gmorph, 2017-05-23:
I can't replicate this.  I have tested on a real rover replicating the scenario and it works for me.  See the base_mode here after I press the safety switch when I have ARMING_REQUIRE set to 0.

2017-05-23 10:22:20.76: HEARTBEAT {type : 10, autopilot : 3, base_mode : 129, custom_mode : 4, system_status : 0, mavlink_version : 3}

Thanks, Grant.

jaxxzer, 2017-05-23:
I did not change the ARMING_REQUIRE parameter. The only parameter I changed from the defaults was ARMING_CHECK to disabled. (and BRD_SAFETYENABLED for the second test).

jaxxzer, 2017-05-23:
Here is where QGC determines the arming status. https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L437

gmorph, 2017-05-23:
@jaxxzer - thanks for clarifying in the Dev Call today.  I can now replicate the issue you are seeing.  I have also tested in Plane and it works fine so this is only an issue for Rover as far as I can tell.  I'll have it sorted shortly.
Thanks, Grant.

jaxxzer, 2017-05-23:
@gmorph I'm glad you've been able to replicate it. Thanks for letting me know!

gmorph, 2017-05-23:
Fixed with this commit and tested on my rover.
https://github.com/ArduPilot/ardupilot/commit/5ac1293a52e007fbd7238e21dae27f5f46d58d4e

Thanks, Grant.

closed 2017-05-23 by gmorph